bri3d/VW_Flash #5: 111SC8E0O300 ASW3 patch compatibilty

hayk57, 2021-01-30:
Hi,

Thanks for your great work, I was looking to flash custom calibration data to a SIMOS18 with the following SW number 5G0906259F__0002 I have decrypted frf, extracted odx and got the 5 blocks. 

My ASW version is O30, I found an address to place the hook it was 0x9474 by comparing the file used in the example (FL_8V0906259H__0001.frf). 

One thing I'm missing is patch compatibility, in a issue opened on 3 Dec 2020 you wrote that if the CBOOT we have contains the bytes DA 00 3C 02 at offset 0x8D1E this will be compatible. I don't have these bytes but even if I look at the CBOOT from the example you provided (8V0906259H__0001) I don't see these bytes at offset 0x8D1.

Could you please give me some additional information to identify if the patch is compatible with my version. 

This is the file I am working on [https://www.dropbox.com/sh/fruy7cak0tkf4p4/AACuqfKUZyRObhKiwafDQ1qva?dl=0 ](url)

Thanks again for this work, I am new to Tricore reverse engineering.

bri3d, 2021-01-30:
Hi! I just wanted to let you know that I've seen your comment and will take a look at your files and see what changes are necessary. It may be a day or two as I have a busy weekend!

I can absolutely confirm that this CBOOT is different from the one used in the patch.bin, so please do not try to use them together. The offsets for all of the functions called by patch.bin (for example, set_task_context and enter_supervisor_mode) are totally different in this CB as well as the location to be patched for Sample Mode / RSA Off. You or I would have to reconstruct the patch entirely by locating each referenced function and changing the offsets as I haven't automated this process at all.

Another option is to flash all blocks from 8V0906259H__0001.frf _and_ the patch.bin at the same time to your ECU. The patch halts execution so early in the ASW startup process that the car won't mind the cross-flash by the time the patch starts. The main reason this is dangerous is that the patch _must_ take for this to be successful. If the unpatched 8V software were to try to finish booting, you could hard-brick because the ImoDat TunIdx for your PClass might not match the calibration.

Either way,at this point with either a working patch on your "target" ASW/CBOOT or a cross-flashed ASW/CBOOT, you'll be stuck in an ASW which instantly boots to an "RSA Off" / "Sample Mode" CBOOT each time it starts.

So the next step is then to patch your target CBOOT in place (i.e. just editing the FD_0 itself to turn on Sample Mode), and flash a complete modified copy of your target SW version.

I hope this explanation makes sense - I recognize my documentation is not very clear for those just looking to flash a CAL, and will also be trying to prepare some more linear documentation for those more interested in the result than the reverse engineering :)

hayk57, 2021-01-30:
Thank you for the quick response and explanations.

I will be very grateful if you can look at the necessary changes, I will also try on my side.

I think that patch reconstruction is the best way to go as I don't want to "hard-brick" the ECU, I will try to load the file, locating the functions with Ghidra and changing the offsets. 

Thanks

fastboatster, 2021-01-31:
+1, I was going through the patch docs and also noticed that FL_8V0906259H__0001.frf I got also doesn't have those bytes at that offset.

bri3d, 2021-01-31:
I will review my notes later tonight to make sure I did not mistype. I
think that this may be the offset into the memory segment that’s loaded
into RAM, rather than the offset into the flash segment.

hayk57, 2021-01-31:
I load CBOOT (FD0) into GHIDRA and the only difference I saw is at offset 0x8001c400 (10310096AA on your file and 10317122AA on my file) the rest of CBOOT is exactly the same. 

I made a mistake comparing FD1 (ASW1) yesterday those are different which has no influence on the patch if I understood correctly. 

So if I place the hook correctly the patch should work.

bri3d, 2021-02-02:
So, I have just made a very worrying discovery. I have found two completely different (but valid, signed!) CBOOT files in different FRF files with the same name, both with specifier 111SC8E0L20 in the header areas. Unfortunately these two CBOOTs have completely different code layouts. The correct version can continue to be distinguished by searching for the CBOOT identification string as returned by the UDS handlers, as I had already documented in "patch.md." This version string is SC8.1 C0 02 SC8 in the version I based my patch on, vs SC8.1 E0 02 SC8 in the other version I have found. 

I will remove the reference to the FRF for now until I can figure out a more consistent way to inform users to source the correct file. Certainly I will post the SHAsum for the correct files to prevent any issues or accidents (before the project is ready for end-user use, I will be sure to have the script itself verify SHAsums to ensure nothing goes wrong).

This is somewhat interesting as some folks have emailed me to report success in working with this file. I'm not sure which flashdaten is which in terms of the file and will have to do some more research here.

A few additional notes about the hook/patch from a few people who have tried different permutations of it out, and some big, huge words of warning:

* The patch is designed to be called from an area with interrupts already disabled, if this is not the case for your hook location you will need to add a `disable` instruction to the patch or you risk some scheduled ASW timer doing something you don't want before CBOOT gets a chance to start.
* There is always a very very high risk of a brick when developing ASW, especially when altering a very low-level startup routine like the one the patch hooks to. If the patch damages the ASW startup in such a way that it fails, you can and will get stuck in an ASW reboot loop which can only be recovered with a commercial tool at this time (using the Service Mode / SBOOT exploit to recover flash passwords and then erasing a validity block to force the ECU back into CBOOT).
* On paper, this entire system could be made a lot safer by hooking the patch-CBOOT-loader to a CAN handler or something set conditionally. But, there's not any unwritten room in any CAN handler to add the patch. Another option is to add the patch-loader to a scheduled task table and have it trigger on some conditional, but by moving the patch-loader to after the ASW boot process, you introduce the Immo to the mix and the correlated IMMO-hell brick risk, and of course if you incorrectly patch the task table you will still crash the ASW anyway. So it is all tradeoffs of non-zero risk.

This is a topic of debate, but here is my two cents: it is possible to get stuck in IMMO-hell (ASW without the ability to enter CBOOT to reprogram - basically an ASW that only allows part number read and passthrough to IMMO for GeKo). This can happen if you cross-flash a full BIN that is calibrated for a different PClass without patching. But, you can also easily brick by just corrupting the ASW startup process with a bad patch.

Therefore, I would argue that without access to bench development tools (boot passwords + skill to enter BSL, or commercial tool with Service/SBOOT exploit), cross-flashing a *known-good* patch with the corresponding ASW is probably the safest approach - that is, something that has been tested on another ECU first, regardless of whether or not it is the same part number.

And in this case, I would love to provide you all with such a file but the presence of multiple CBOOT files for the same version specifier throws an exceptionally large wrench in the operation that I did not anticipate.

fastboatster, 2021-02-02:
Well, just as I was reading about the risks of patching low-level ASW startup functions I was going to suggest one of the CAN handlers but then you explained why this was not viable in the next paragraph. Kind of sucks. But anyways, I think you have already suggested before that people just need to flash the stock file you developed the patch with and go from there. So having different CBOOTs is not an issue as long as we do as you suggested. Also, I'm currently watching your work hoping that some Simos ecus older than 12 such as 8.4/8.5 will be added eventually. It seems to me, they have the same vulnerability which allows to write to an unerased block and also don't even use aes128 to encrypt the payloads, although these ecus might be of interest to a very few people and there are already some relatively inexpensive proprietary tools to write to these ecus. But could be another low-hanging fruit and open source ftw. So if there are any plans to look into these ecus, I'd like to help to the extent of my (rather limited) abilities. The big thing which stops me from trying to modify and flash the asw is bricking the ecu of course and I haven't figured out how to flash an extra ecu outside of a car.

bri3d, 2021-02-02:
I don't have an old Simos but I could probably reverse a few files and add some attempted support if it is similar. If there is no AES, I will have to adjust my flasher a bit to not request encryption when it writes for that ECU series. 

Yes, the idea was that just one patch would be enough as people could flash a single firmware and patch, and since the patch hooks in so early they wouldn't need to worry (too much!) about getting stuck in IMMO hell due to PClass. Unfortunately learning today that there seem to be multiple CBoots associated with the same software makes this a lot harder, because I now need to figure out which file is which to tell people which one to get, or re-create my patch on a software version which is more consistent and doesn't have this problem (I may simply work on the latter and shift to something like the 259K__003 file which seems fairly common, as long as I can make sure there aren't multiple CB versions hiding there too...). 

I have been spending most of my time recently on reversing SBOOT with the hope of finding the Service Mode exploit in S18. This will allow the use of junkyard ECUs for further development without needing very expensive commercial tools, as the Tricore password can be recovered and then the IMMO function which is referenced by UDS can be patched out by manually overwriting the flash pages using BSL.

fastboatster, 2021-02-02:
Thanks, appreciate this! I'm going to say I learned A LOT from your repos and forum posts so thanks a lot for this, too! Just by looking at VW_Flash/Simos 18 things like checksums and memory addresses, I was able to quickly identify the checksum locations in Simos8.4, find out memory addresses for CBOOT, CAL, and ASW. With that, I was able to quickly find a few direct references to some calibration maps such as torque limits by gear and rpm and what looked like a lookup function that fetches values from that map based on a pair of indices (probably rpms and gear). Anyways, I've been digging in simos 8.4 disassembly for some time now, obviously wasn't overly successful. It doesn't seem to be very different from 18 except that it doesn't use lzss either, it's just an xor compression like with Simos 10. I'm trying to find a func that descends in the cboot from the ASW and copies some cboot parts into the RAM.

bri3d, 2021-03-06:
Hi, I have updated the patch binary and docs with a tested patch for the correct, non-tampered `E0` CBOOT from H__001.frf . Apologies for the confusion and please see if it makes sense and matches the addresses in your files.

I also added some additional documentation about back-influencing the CRC properly in the patching process.

closed 2021-03-06 by bri3d