bri3d/VW_Flash #10: Clear DTC after flash doesn't work on windows

joeFischetti, 2021-03-06:
There's a send_obd call at the end of the flash process... but send_obd doesn't work with tactrix since it tries to make a new connection to the DLL when there's already an existing one opened....
One fix would be to call a close on conn BEFORE calling send_obd (so that the dll gets closed, and conn2 can re-open it with a new rxid/txid.
Not sure how that would play with the other interfaces though.  The clear dtc is the last step in the process, is there any reason not to close them?

bri3d, 2021-03-06:
That seems reasonable and cleaner than the current system, feel free to make the change or I'll try to get to it. It might even make that part of the process more reliable.

joeFischetti, 2021-03-06:
Just checked - that wont work (with J2534 anyway) because there's an additional `tester_present` sent after the final clear dtc in `simos_uds`.  

J2534 can 'reset the cable' with new txid/rxid's though I haven't actually tested that part of it.  it's an option for sending the 0x04 to clear dtc's from the existing `conn` (instead of recreating `conn2`)

Alternately, we could `conn.close()`, `send_obd()`, and then `conn.open()` again, though that means I actually need to rewrite `J2534Conncetion.open()` since it does *not* connect to the dll (that's done during init currently)

Is that final `tester_present` necessary?  line 505, before the final callback.

bri3d, 2021-03-06:
No, I don't think it should be. Technically nothing after checking the final programming conditions is "necessary" to produce a booting ECU, although clearing the DTCs is certainly nice for the user. That tester_present check was there when this was a development tool, so I could easily see if the ECU came back up into ASW properly. Now that we're more focused on user experience, it isn't useful anymore so we can remove it.

bri3d, 2022-01-25:
As we know this is still kinda unreliable, but we did fix the mentioned issue with a TesterPresent after the send_obd causing j2534 issues, so I'm closing this old one out.