bri3d/VW_Flash #11: Implement boot password read/store during flashes

joeFischetti, 2021-03-06:
For obvious reasons, we should archive boot passwords as often as possible (not just on full flash).
Mainly because you may be using a different machine or your original backup from you full flash is unavailable.

bri3d, 2021-03-06:
This is probably not very feasible using our current patch tooling. We would need to add an unrestricted read primitive to each CBOOT or ASW the end-user is running in order to do this on every flash, which means effectively would mean writing a working "intelligent" patcher which worked off of disassembly to ensure compatibility with multiple passed in files. Lots and lots of work if we want to make it cross-compatible.

It's a lot easier to just add one handler to the "initial patch" and store off the passwords only once, or force everyone to use the same software version. 

I do have adding boot password extraction to the initial ASW->CBOOT patch as an action item, but I don't know it's practical to do this on every flash.

joeFischetti, 2021-05-12:
Closing this
With the SBOOT bench exploit there's really no concern over whether the boot passwords are available - juice isn't worth the squeeze.

closed 2021-05-12 by joeFischetti