bri3d/VW_Flash #16: Data padding to 16 bytes issue

dirkcgrunwald, 2021-05-20:
Hi,

I'm trying to learn more about the ODX format and conventions for VW variants of the files. I don't have access to the ASAM specifications but I do have a set of three ODX files from VW.

When I try to run the "extractodxsimos18.py' script, I initially got failures & inserted some debugging info as below:

```
python3 ../VW_Flash/extractodxsimos18.py --file FL_03L906012BM_4824.odx --outdir foo
length of dataBinary is 41469
key is  b'\x98\xd3\x12\x02\xe4\x8e8T\xf2\xcaV\x15E\xbao/'
Failed to decrypt, exception is  Data must be padded to 16 byte boundary in CBC mode
length of dataBinary is 1854128
key is  b'\x98\xd3\x12\x02\xe4\x8e8T\xf2\xcaV\x15E\xbao/'
Failed to decrypt, exception is  bytearray index out of range
length of dataBinary is 495
key is  b'\x98\xd3\x12\x02\xe4\x8e8T\xf2\xcaV\x15E\xbao/'
Failed to decrypt, exception is  Data must be padded to 16 byte boundary in CBC mode
length of dataBinary is 2414
key is  b'\x98\xd3\x12\x02\xe4\x8e8T\xf2\xcaV\x15E\xbao/'
Failed to decrypt, exception is  Data must be padded to 16 byte boundary in CBC mode
length of dataBinary is 510082
key is  b'\x98\xd3\x12\x02\xe4\x8e8T\xf2\xcaV\x15E\xbao/'
Failed to decrypt, exception is  Data must be padded to 16 byte boundary in CBC mode
```

It's throwing an exception at the `cipher.decrypt` line when invoking the AES CBC cipher. All three fail the same way.

I'm very certain these ODX files are valid but don't know enough about the specific format details to figure out what's wrong. I've looked at e.g. the "preview" at https://www.sis.se/api/document/preview/910417/ but can't find details on the <FLASHDATA> assumptions.

I do note that while the compressed data is not a multiple of 16 bytes, the uncompress block is a multiple of 16.
[image]
Is it supposed to be uncompressed and then decrypted?

Is there more publicly available info on the format? Happy to send a PR if/when I get this figured out.

bri3d, 2021-05-20:
> find details on the assumptions.

Hi,

ODX is a standard format which can apply to any flash payload, and
unfortunately my decompression and decryption script only works on payloads
for Simos18.1/Simos18.6 with Encryption A (“Audi method” - fixed key/IV
AES) and Compression A (“Simos compression” - LZSS).

Other ECUs using the ODX container will have different encryption and
compression methods, and even “encryption A” can differ from ECU to ECU
(for example the AES keys on Simos18.10 are different).

The only good way I am aware of to find out these formats is to disassemble
a decrypted ECU binary and go hunting - this is how I found the Simos18.6
keys.

bri3d, 2021-06-16:
Closing this one out since it's not really an issue in the decompressor/decryptor.

Please let me know if my information was helpful. That control module looks like EDC17?

I don't do a lot of research into them, but usually these older Bosch control modules used some form of primitive XOR scrambling, sometimes just rolling-XOR with a cutesy fixed code or 255-byte key. It's hard to say with your specific module but that's where I'd start. Cryptanalysis (try to find repeating/known data, xor backwards) or disassembling the flash handler are usually the best bet for figuring these out, or trying to find them documented elsewhere.

closed 2021-06-16 by bri3d