bri3d/VW_Flash #24: Force controller choice when run through CLI

joeFischetti, 2021-09-07:
Since the project is getting larger and expanding to additional controllers (TCU, haldex), instead of doing --dsg, we should do --controller=dsg (and similarly, --controller=ECU).  Defaults are good, but I wouldn't want to flash_bin my ECU with a TCU driver. 

It's probably also a good opportunity to work in some additional idiot proofing (like flash_cal does) to at least prompt the user before potential crossflashing.

bri3d, 2021-09-07:
agree that explicitly selecting a controller makes sense, will update to add. hope nobody had this in automation scripts since it's a breaking change.

joeFischetti, 2021-09-07:
Good point.
Then again, it's not like we're not breaking various other parts of the
project on a regular basis.