bri3d/VW_Flash #26: Could this be used to flash virtual cocpit?

OneB1t, 2021-09-27:


bri3d, 2021-09-27:
Not sure. Each control module requires its own reverse engineering effort.

Quickly looking at an ODX for these units:

* Flashing most control units using a stock, signed FRF is pretty easy. This unit seems to use the normal `FlashJobUDS` layer so it may be possible to flash an FRF with some enhancements/patching (although I see some unimplemented signature blocks in there which we currently do not send with the tester - would need to look and see if these need to be sent or not). 

* I have no way of knowing whether the control unit will allow cross-flashing or what will happen if it is attempted, as I don't own one. The data in the ODX seems to be encrypted so it would take some reverse engineering effort to pull it apart and find out what is going on that way, too. 

In short:

* No current support.
* It would probably be possible to add FRF flashing support for this control unit with small to moderate effort, although without owning the module I can't be too sure.
* I can't tell you whether or not the unit will allow a cross flash as-is without more information. Can this be done with ODIS Engineering or VCP?
* If it doesn't allow a cross flash as-is, I can't tell you how hard it would be to patch it to allow a cross-flash as this would be its own reverse engineering effort.

OneB1t, 2021-09-27:
Thanks for info i just ordered that cluster and it will be my winter project to make it work in my car somehow :) so im looking for more informations how to hack it

OneB1t, 2021-10-04:
i want to unpack ODX file to get binary so i can poke it via ghidra or some other tool..

and im stuck on following part
[image]
this part inside ODX is have following compression/encryption
              <ENCRYPT-COMPRESS-METHOD TYPE="A_BYTEFIELD">AA</ENCRYPT-COMPRESS-METHOD>
where should i get AES key and IV part? is it also somewhere inside ODX or only way how to obtain this is from firmware binary?
SA2 i can see inside ODX so that i can reuse

bri3d, 2021-10-04:
The `ENCRYPT-COMPRESS-METHOD` in these Virtual Cockpit ODX files is actually `1A` . 

There is no guaranteed consistency for encrypt/compress methods across control units. It's simply an identifier which is sent with the RequestDownload message to the control unit to let it know what data to expect.

So, there is no way to say for sure that the encryption is AES, or what it is at all.

You have a few options here:

* Open up a control unit and see if an unencrypted firmware can be extracted physically (debug interface, ROM chip, serial interface, etc.).
* Hunt through internal / engineering file leaks ("map packs," seedy forums, etc.) to see if an unencrypted ODX file or some kind of engineering firmware dump exists in one.
* Enter a programming session and test if RequestUpload was left in UDS for the control unit (unlikely, have not seen it implemented for years).
* Extract the encrypted/compressed ODX data (edit the python script to write out the dataBinary directly) and attempt some form of statistical cryptanalysis on the encrypted/compressed data. This is made a lot harder with compression, but for example on old control units which just used an XOR key, the key could be recovered this way.

OneB1t, 2021-10-05:
i think there is NAND flash 29F4G08ABADA inside cluster today i split it open this is how it looks like 
[image]

does it make sense to try to read/write it directly? or there will be some other catch :-/

bri3d, 2021-10-07:
Odds of the NAND being encrypted are pretty low I think. Dumping it would definitely be a great start.

bri3d, 2021-10-08:
Reading TSOP flash chips is pretty standard. Your list is fine to start, although to put it back you probably want solder paste instead of just flux.

For solder/desolder you could also use ChipQuik instead of hot air if you want. Or, look into a TSOP clip like 360clip. 

> also when ODIS is flashing firmware is there some part where I can sniff for unpacked raw binary (maybe after CAN->serial converter) or it is decrypted on cluster itself?

No, the encryption/decryption is done in the cluster itself, this is the fundamental protection method in most newer control modules - ODIS just reads the ODX and does what it says, the encryption and compression is performed in the control module.

> EDIT2: what is purpose of 8 pin "debugging interface" which is accessible on back side of cluster?

With respect to both the debug interface and QEmu, maybe, you'll need to figure out what the CPU is (V850)? And go from there.

OneB1t, 2022-03-10:
ok so managed to dump that NAND flash but it does not look much useful as there is no strings which i can use to make navigation in ghidra easier

is there some working tutorial i can use to run it in qemu for example?

markis90210, 2022-05-16:
Hello. Maybe someone sucessfluly flashed virtual cockpit by mib2?

OneB1t, 2022-05-16:
There is update firmware for infotainment SD card on mib solution https://mibsolution.one/

markis90210, 2022-05-17:
> There is update firmware for infotainment SD card on mib solution https://mibsolution.one/
> 
> Also what is cheapest way to obtain "mqb testrig"?

Make it. Few wires, connectors, getaway, few displays, depends by unit, and few keyboards. Also usb port.

OneB1t, 2022-05-17:
But this is probably not enough as it will require also working key + immo in order to flash firmware into cluster right?

To just wake cluster up i can probably use just Arduino + CAN shield

But I'm interested in "minimum viable rig" to flash VC on bench

markis90210, 2022-05-17:
> But this is probably not enough as it will require also working key + immo in order to flash firmware into cluster right?
> 
> To just wake cluster up i can probably use just Arduino + CAN shield
> 
> But I'm interested in "minimum viable rig" to flash VC on bench

Can we communicate in messages? Skyle, whatsapp, viber, telegram?

OneB1t, 2022-05-17:
Sure we can. Check xda-devs ;-)

markis90210, 2022-05-17:
> Sure we can. Check xda-devs ;-)
Write to you on skype :)

heeyroen752, 2022-11-09:
Did you already had some progress with crossflashing the virtual cockpit? I am a mechanical engineer and want to learn these kind of interesting programming. Specially because I do many “programmings” (odis) in these vag cars. I heard about modified/patched firmwares for the virtual cockpits which are being used to crossflash/cp off etc

markis90210, 2022-11-10:
Hello. Unfortunately no progress.

OneB1t, 2022-11-10:
Unfortunately there is not much research going on for these devices. Mine is still waiting on the desk for more public information's :-)

markis90210, 2022-11-10:
But  now it is hard times, because dealers not accept to adapt used
cvirtual cockpits, so i think smart people start search ways how to adopt
it without dealer, and ofcourse what will find whe way. I was send one
cluster to people who unlock processor, to possibility to change mileage. I
Was thinking, that after unlock, will be possible to read procesor, but
unfortunately no. They unlock something only for mileage correction. And i
see, that they unsloder all processor, not connect by processor pins

OneB1t, 2022-11-10:
I can read that NAND flash but it is either encrypted or something weird is going on there. Without some internal knowledge this will be tough challenge.
It's interesting that some company is able to mess with odometer now, maybe there is still hope :-)

heeyroen752, 2022-11-10:
So what i know is the data from mileage, sport layout cp and some of them are being locked if the dash is over 100km. This data is being unlocked 2 ways. 1 by unlocking flash and second one by your hacked firmware. Same way they hack the mib2 system. I even readed some people have this solution by just sending can commands. Found some kind of patched firmware but not used yet as i dont know what the background/information is and what it does

OneB1t, 2022-11-11:
Can you share this hacked firmware? ODX files are encrypted and probably signed but maybe we can see what they did in order to go around these limitations :-)

markis90210, 2022-11-11:
About what firmware you are talking?

heeyroen752, 2022-11-11:
Yes ofcourse i can share it. How can i share it with you? You have telegram or something? The file is not odx. It is updated by sd through mib2 (most)

OneB1t, 2022-11-11:
maybe upload it to mibsolution? ;) https://mibsolution.one/

heeyroen752, 2022-11-11:
Oke i will try today afternoon. Only thing is i don't think everyone in mibsolution admin group should be pleasured. Some want solutions not too be public as you can also see in the telegram group of them

OneB1t, 2022-11-11:
I don't think that flashing firmware should be some "underground operation". On many other platforms phones, pc or even my dust cleaner it is routine operation. But for some reason in automobile industry you need to be some sort of shaman....

heeyroen752, 2022-11-11:
https://drive.google.com/file/d/1SN4dpzfOhfeUcmpOY3xjDinIQkIeHk3h/view

OneB1t, 2022-11-14:
Do you also have unmodified version of this "update"?

heeyroen752, 2022-11-15:
Unfortunately not i will check if i can find it

OneB1t, 2022-11-15:
This looks like update for AUDI cluster. I'm not sure if that is also applicable for VW/SKODA clusters as there is some files for NVIDIA processor which I believe is not used inside 5NA920790B or similar

qnxvag, 2022-11-20:
A few notes based on my experience with Gen1 VC units ie  TT and Q7 4M0 (I didn't get past accessing KSS):

1. KSS = Atlas Chip MB9DF125 (this is a specific chip designed for automotive industry for VC. It has a secure internal 48KB EEFLash in it. Mileage and other "special stuff" stored here). @Markis90210 - which chip did they desolder? MB9DF125?
2. 2. GSS = Nvidia board running QNX. It talks to KSS via the KSS Driver. 
3. Current JTAG attempts to access KSS have failed (that I am aware of). If JTAG was possible then 
4. NAND is just the storage for the Nvidia QNX board. So it contains boot partition information and QNX partitions etc..
5. If you flash the NAND from another "version" you can kill the VC as there is a "version check" between the KSS and GSS during bootup. They need to match the versions in the update media... 

[image]

Hopefully this information helps 
[image]

OneB1t, 2022-11-21:
Is there a way to obtain console? There are some root passwords for this gen1 vc https://mibwiki.one/doc/mmx-root-passwords-Ksdm2xg8md

qnxvag, 2022-11-21:
yes supposedly... there is an 10 pin 0.5mm pitch zif connector with no zif cable connected to it..  You need to open up the VC unit to access it. Some development units had the air grille on the back with some grilles removed so a cable could be run from the connector to a bench setup.... You need to figure out the pins for ttl and USB. 115200... etc.. etc..

OneB1t, 2022-11-21:
ok any idea where JTAG/serial is on my type of VC? As it looks like there is quite a lot of info about these NVIDIA tegra based AUDI VCs but basically nothing about my device :-(

I think that for AUDI VCs mileage correction will be possible by just messing with QNX jar file (take readout from security chip and just +- some value to it :-D)

qnxvag, 2022-11-21:
People are offering to remotely update sports layout on the Gen1 VCs and
VAG have "x" versions available for purchase which supposedly are
"refurbished" units so they definitely have a way to do it.

Are there are any zif connectors on your test unit that are not used? there
must be a way to access the board..

OneB1t, 2022-11-21:
This is how full PCB looks like any idea where to start search for console?
[image]
[image]

qnxvag, 2022-11-21:
are all the connectors connected to something? If not, then concentrate on
that connector. If not, then you need to probe solder pads with an
oscilloscope to try and find low voltage signals..

heeyroen752, 2022-11-23:
This is what i have

[image]

qnxvag, 2022-11-23:
yep, Gen1 pinouts from what I know... Gen2 is obviously a different beast

heeyroen752, 2022-12-06:
By the way. People are doing this without opening the cluster. The solutions they have is by flashing the unit by mib2 (most). It seems they used patched flash files which unlocks the secured area for the special stuff (mileage, sport layout, wheel circumference etc)

kurchan, 2023-01-16:
Hello everyone Today I received a panel from audi Q8 (Virtual) on which they reset my mileage to 0! there was nothing soldered in the panel. Question??? Where can I get patched firmware???

heeyroen752, 2023-01-17:
Unfortunately not a public solution. There seems to be a possibility without even using a firmware update. They use uds commands to unlock the locked adaptation values. I wished i could have these information as well :)

arthurdurlesteau, 2023-01-23:
Hello, maybe somebody can help me to remove CP from virtual cockpit? Odis doesn't work in my case but hear that it is possible to do offline.

ranger0909, 2023-02-05:
> Hello everyone Today I received a panel from audi Q8 (Virtual) on which they reset my mileage to 0! there was nothing soldered in the panel. Question??? Where can I get patched firmware???

Please can you give me the contact to reset please?thanks

kurchan, 2023-02-14:
skype: kurchanov

bri3d, 2023-02-15:
VW_Flash issue comments are not a file trading website and not a place to discuss selling solutions.

VW_Flash is a good starting place to understand SA2 authentication, UDS flashing, and basic bootloader structure and exploit development, but once you start getting into individual exploit development, you should find a forum that's better suited than GitHub Issues, and "dm for solution" is not in the spirit or goals of this project.

I will leave most of the comments in this thread up as they represent good research, which I am not opposed to in any way (I think it just needs a better place to happen).

However, I am removing paid / secret solution related comments as they are not in the spirit of this project at all.

Basic UDS reflashing should not be a mystery to anyone at this point. , Programming Preconditions, Enter Reflashing/Bootloader Session, SA2 Security Access, Erase, RequestUpload, TransferData, ExitTransfer, Checksum, Repeat, Programming Complete. Everything beyond the basic routines is going to revolve around engineered exploits specific to the code running on an exact control module, not some generic "reflashing secret" that nobody is letting you in on.

dashcoder13, 2023-02-18:
Hello guys ,

i want to know if someone have solution to set mileage on audi vc after retrofit .  I want to set original mileage of the car .

OneB1t, 2023-02-18:
It looks like there will be abrites interface which will be able to do mileage correction looking forward to that 

I never really moved anywhere  after NAND read and there is very little about components in VC so really not much to build on...

dashcoder13, 2023-02-18:
> It looks like there will be abrites interface which will be able to do mileage correction looking forward to that
> 
> I never really moved anywhere after NAND read and there is very little about components in VC so really not much to build on...

Hi ,

abrites will not release programmer for audi virtual its only for renesas mcu . Not for fujitsu .

ranger0909, 2023-02-19:
Hello you must not dream if you read the arm chip mb9df125e meter die the hw and block that is why it uses the mib

All vag vc virtual is easy is not same audi

OneB1t, 2023-02-19:
can you explain more what tools / setup is required for normal VAG cluster (like 5NA920790B)?

there is only very little info on how to work on these clusters (JTAG, how to wake them up, some datasheets, what is stored where etc..) i have it on desk but i just dont know how to continue :-/

dashcoder13, 2023-02-19:
Hello 

poldiag car read vw virtual cluster.
I use it to do the job .
https://keymaster.pl/shop/?v=9b7d173b068d

Best Regards

dashcoder13, 2023-02-19:
> Hello you must not dream if you read the arm chip mb9df125e meter die the hw and block that is why it uses the mib
> 
> All vag vc virtual is easy is not same audi

@ranger0909 do you have skype or whatsapp to talk .

dashcoder13, 2023-02-19:
@bri3d mate can we talk in private i have odx file for vc

procarcyprus, 2023-02-22:
> VW_Flash issue comments are not a file trading website and not a place to discuss selling solutions.
>
> VW_Flash is a good starting place to understand SA2 authentication, UDS flashing, and basic [...]

hi boss how to contact u this side private ?

procarcyprus, 2023-02-22:
> @bri3d mate can we talk in private i have odx file for vc

boss pls send me private msj i need odx file virtualdash board

heeyroen752, 2023-02-22:
> @bri3d mate can we talk in private i have odx file for vc

Can you send me this odx file

procarcyprus, 2023-02-22:
hi boss do u have whatsupp or skype

heeyroen752 ***@***.***>, 22 Şub 2023 Çar, 19:29 tarihinde
şunu yazdı:

heeyroen752, 2023-02-22:
> hi boss do u have whatsupp or skype
> 
> heeyroen752 ***@***.***>, 22 Şub 2023 Çar, 19:29 tarihinde
> şunu yazdı:
> 
> > @bri3d <https://github.com/bri3d> mate can we talk in private i have odx
> > file for vc
> >
> > Can you send me this odx file
> >
> > <https://github.com/bri3d/VW_Flash/issues/26#issuecomment-1440362371>, or
> > unsubscribe

Skype live:intumaco_h

[1 more reply]