bri3d/VW_Flash #29: Trying on windows with j2534

nickGaiot, 2021-11-17:
[t.write(did + " : " + ecu_info[did]) for did in ecu_info]
TypeError: 'NoneType' object is not iterable

bri3d, 2021-11-17:
Please take a look at `flash.log` and `udsoncan.log` for more information. It's also better if you post the command line you are using in bug reports as well, although in this case I was able to infer what you were trying to do.

It seems that your ECU unexpectedly sent a negative response code to one or more ECU info DID requests, or there was an issue with the J2534 stack which prevented reading a response.

nickGaiot, 2021-11-17:
Sorry, i'm tryin g to perform this command : 
py VW_Flash.py --frf frf/ciao.frf --action flash_bin --dsg --interface J2534
And looking in the flash.log I have this error : 
2021-11-17 20:00:03 [ERROR] SimosFlashHistory: Service request timed out! : TimeoutException('Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)')

bri3d, 2021-11-17:
In the udsoncan.log, what was sent to the DSG before it timed out? If you can send me the full logs (via email perhaps, you can find it on GitHub) I can probably help to debug more. 

A little odd here as flash_bin shouldn't be triggering any code that looks like that, at least to my understanding - that looks like the code from `get_ecu_info` which will probably request items that aren't supported by DSG.

nickGaiot, 2021-11-17:
Sorry for my confusion, I'll explain :
The first " [t.write(did + " : " + ecu_info[did]) for did in ecu_info]" error was trying to get_ecu_info without any flags..  but anyway, my main purpuse is to remap my Dsg not my Ecu and im trying using `py VW_Flash.py --frf frf/ciao.frf --action flash_bin --dsg --interface J2534`
And i have these logs
[flash.log](https://github.com/bri3d/VW_Flash/files/7557802/flash.log)
[flash_details.log](https://github.com/bri3d/VW_Flash/files/7557803/flash_details.log)
 
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/7557805/udsoncan.log)
:

bri3d, 2021-11-17:
32bit Python has been tested as it is indeed the only way to have OpenPort work (since the DLL is 32-bit), so it's not causing the issue for you. 

Yes, connected to OBD port on an 8V S3 with the ignition/dashboard on but engine not running should be correct.

For whatever reason no response is making it back at all to any request.

You were on the right track testing with `get_ecu_info` with no flags - I think trying different things until that works is the right approach, and then once you can talk to the ECU, try talking to the DSG.

Based on the logs, I don't know what's going on. It seems that the OpenPort DLL is being loaded and is acting as though it is sending a message - just, nothing is ever received. Usually this is just a symptom of the car ignition not being turned on, but it could be an issue with the OpenPort or drivers as well. 

We have had issues with some clone OpenPort devices and not others, which could play a role too. 

One thing I did notice is your OpenPort driver version seems different from the latest one that most users are on. I looked at a log from a successful use of OpenPort on Windows and it had the following:

`FirmwareVersion: b'1.17.4877', dllVersoin: b'1.02.4820 Jul  6 2016 17:20:04'`

Maybe try installing a newer OpenPort DLL driver package?

nickGaiot, 2021-11-18:
Thank you the advice, I'm using an openport clone from China, now I installed the official drivers and the versions is this one: `J2534 FirmwareVersion: b'1.17.4877', dllVersoin: b'1.02.4791 Jun  2 2016 12:43:17', apiVersionb'04.04'` but same Issue .
How can I know if my clone is not supported? i bought it for  40€, And official dll should work with clones?
Thanks for your Time

bri3d, 2021-11-18:
I have no idea how to determine if your clone is supported or not, honestly.

After checking that the ignition really was on, if I were tasked with debugging this I would use a OBD port splitter and a CAN sniffing tool to see if the clone device was actually sending data, if the ECU was actually responding, and if the data made it back. Then I would use a USB sniffing tool to see if the issue was in the device firmware (data made it through USB bulk endpoints but not to app layer) or the DLL.

But, I don't much care for figuring out clone tools. I will say that some end users have been successful using OpenPort devices with printed serial 1234, a clear port connector, and the factory OpenPort DLLs, but of course there are probably a range of clones and clone firmwares out there all with their own weird quirks, this is after all the problem with clones. Devices with a black port connector seem to have less reliability.

nickGaiot, 2022-01-22:
Hi, I tried to troubleshoot the j25434 clone but with no luck , so I decided to give it a try with the rasberry with can module ,I used the cheap can Module MPC2515 modified to work with rasberry and after some errors I managed to make it work! now I can perfectly comunicate with the ecu and the dsg , I flashed my DSG with custom calibration and it worked, thanks for the amazing project , and thanks again for your time.

bri3d, 2022-01-25:
Thanks, I am glad it worked. Sorry you couldn't get your clone working - some seem to work just fine and others don't.

I'll go ahead and close this out (it will still be in the Issues archive for folks to find).

closed 2022-01-25 by bri3d