bri3d/VW_Flash #34: Aptempting to flash Simos 12 or Extract FRF

nickGaiot, 2022-01-31:
Hi, today first I tried to get bins out of my `fl_8v0906259k__0003.frf ` FRF but i get an error that I also got when I tried to flash the FRF to the car the line that I ran is `sudo python VW_Flash.py --action flash_frf --simos12 --frf frf/FL_8V0906259K__0003.frf `  when trying without the --simos12 flag it works, maybe is only and issue with the simos 12 part .
[image]
[flash_details.log](https://github.com/bri3d/VW_Flash/files/7974400/flash_details.log)
[flash.log](https://github.com/bri3d/VW_Flash/files/7974401/flash.log)
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/7974402/udsoncan.log)

bri3d, 2022-02-01:
FL_8V0906259K__0003 is Simos18 software, so it doesn't work with the Simos12 switch since the AES keys are different.

nickGaiot, 2022-02-01:
Thanks, after I posted this I discovered that is for Simos 18 now I have further questions, but I  wrote you an email because are not exactly an issue .

closed 2022-02-01 by nickGaiot