bri3d/VW_Flash #35: DSG DQ381

MaddinMK7R, 2022-02-05:
Is it possible to flash a DSG DQ381 in Golf7R with the DQ250 function? Identification of DQ381 was successful!

bri3d, 2022-02-06:
Untested and it is a totally different control module with totally different code (as it is Bosch so they reused their existing OS tools so it's similar to Bosch ECU, vs DQ250 which is Temic).

I can look into it but not likely in current form.

MaddinMK7R, 2022-02-06:
That would be great if you could check that. I could give you a log with ECU information and an original .frf file from my DQ381. I would also agree to test on my vehicle.

bri3d, 2022-06-07:
I am attempting to implement DQ381 support. Right now it doesn't look like DQ381 has any RSA signatures or specialty checksums, just simple CRC32. The flash payload is encrypted using AES with a trivially silly key/IV, and compressed using what looks like standard LZSS.

I have implemented checksumming and what should be the correct basic flashing procedure.

It's ready for testing, but **only by someone with another flashing tool which they can use to recover their control unit if the procedure fails.**

MaddinMK7R, 2022-06-11:
I wanted to test it today, I always get this error message, the same when trying to flash simos with this master version. No problems with the release version.

dashing module Abt loaded
2022-06-11 21:37:37 [ERROR] VWFlash: No 32534 DLLs found in HKLM PassThrusupport. Continuing anyway
2022-06-11 21:37:42 [ERROR] VWFlash: No 32534 DLLs found in HKLM PassThrusupport. Continuing anyway.
2022-06-11 21:37:50 (INFO] VWFlash: User selected: USBISOTP_ COM4
2022-06-11 21:41:08 [CRITICAL] VWFlash: Selected: C: \Users \T410 \Desktop\GOLF_7.5_R\D0381\ Tune\eGC308012A_1626_0cl
1 G7R Test MaMo_ 550Nm) - ) .bin
2022-06-11 21:41:08 [WARNING] VWFlash: Keeping block FD_©1DATA because it has no version specifier.
2022-06-11 21:41:08 [WARNING] VWFlash: Keeping block FD_O2DATA because it has no version specifier.
2022-06-11 21:41:08 [WARNING] VWFlash: Keeping block FD_O3DATA because it has no version specifier.
Exception in thread Thread-4:
Traceback (most recent call last) :
File
"C: \Users\T410\ AppData\ Local \Programs \Python\Python39-32\lib\threading.py"
, line 980, in
_bootstrap_inner
self.run()
File
"C: \Users\T410\AppData\Local\Programs\Python\Python39-32\lib\threading.py",line917,inrun
self._target(*self._args,
**self.
_Kwargs)
File
"C: \Users\T410\Desktop\VW_Flash-master\VW_Flash-master\lib\dq381_flash_utils.py",line234,inflash_bin
prepared _blocks
= prepare_blocks(
File
"C: \Users\T410\Desktop\VW_Flash-master\VW_Flash-master\lib\dq381_flash_utils.py",line102,inprepare_blo
compressed_binary=lzss.lzss_compress(binary_data)
File
"C: \Users\T410 \Desktop\VW_Flash-master \VW_Flash-master\lib\lzss_helper.py",line 22, in lzss_compress
p
= subprocess.run(
File
"C: \Users\T410\AppData\Local\Programs\Python\Python39-32\lib\subprocess.py",line505,inrun
with Popen(*popenargs, **kwargs) as process:
File
"C: \Users\T410\AppData\Local \Programs \Python \Python39-32\lib\subprocess.py"
, line 951, in
init
self.
•_execute_child(args, executable, preexec_fn, close_fds,
File
"C: \Users\T410\AppData\Local\Programs\Python\Python39-32\lib\subprocess.py"
, line 1420, in
_execute child
hp, ht, pid, tid =
winapi.CreateProcess(executable,args,
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden

bri3d, 2022-06-11:
It’s missing lzss.exe. the automated build process should have made it. You
can build it or copy it from an older version as it did not change. I am
away for the weekend.

bri3d, 2022-06-14:
As an update:

* I do not need additional testing for now. I did an initial round of testing and discovered that DQ381 requires a very specific modification to LZSS compression, where the compressed data is made to decompress to a precise length. This is actually not a particularly easy thing to do due to the way LZSS works, so I am working on finishing up my implementation of this exact length compression today.

I will push a new prerelease version 0.3.2 or higher once I have solved this LZSS problem.

Progress is being made!

MaddinMK7R, 2022-06-22:
has anyone tested the beta 0.3.4 with a tuning dsg file?
Is the tuning file flashed with the function Flash Stock or Full Flash Unlocked ?

bri3d, 2022-06-22:
It doesn't work yet.

I fixed the LZSS issue, so that part is solved. Now you can use Flash Calibration to flash Block 3 (calibration) if you have a matching ASW already, but Full Flash Unlocked does not work because of an ISO-TP Continue message / timing issue with sending the Erase command for Block 2.

This is very hard to bypass for Tactrix cables (I think this is why commercial tools do not support this module on Tactrix, I now understand...) and for the A0, I have been working on patches to the firmware but we haven't had success yet.

Progress is slow since I rely on others to test.

sagishm, 2022-06-22:
> It doesn't work yet.
>
> I fixed the LZSS issue, so that part is solved. Now you can use Flash Calibration to flash Block 3 (calibration) if you have a matching ASW already, but Full Flash Unlocked [...]

Pcmflash have protocol for obd with Tactrix…

MaddinMK7R, 2022-06-24:
It would be great if the exact parameter set designation-data set name (example: FL_0GC300012A_1626_OcWY.par) were also read out with the TCU identifier, then you would have the exact designation of the required file and you save additional reading with another tool.

KRACER666, 2023-07-03:
Is it possible to write to DQ381/DQ380 now? Will it cause the gearbox to brick?

closed 2023-12-31 by bri3d