bri3d/VW_Flash #36: Enhancement idea for tool suite - raw UDS shell

tinytuning, 2022-02-08:
Not an issue or a fault, but a nice-to-have additional tool in the toolbox.

Sometimes it’s quite useful to be able to manually construct a specific sequence of hex bytes, send it across to the ECU and observe what comes back. This is about passing an ad hoc, user-specified byte string to the underlying ISO-TP encoder/decoder already present within VW_Flash, especially since VW_Flash already contains support for multiple hardware options and is handling the stack. The interface could be a simple shell, invoked from within VW_Flash, that would allow you to type in the bytes you want to send. All the ISO-TP segmentation would be taken care of by the stack, you are simply entering some bytes and getting a response in return. In essence, this gives users the ability to create and send their own UDS commands, assembling the commands byte by byte. 

Pseudo listing below, of how this could be potentially invoked and used.

```
python3 VW_Flash.py --action UDS_Shell
Welcome to the raw UDS shell. Type help or ? to list commands, Ctrl+] to quit.
(raw UDS)>

```
Example #1 – Reset DTC
```
(raw UDS)> send 0x04
received 0x7F 0x04 0x78
received 0x44

```
Example #2 – Set Long Coding
```
(raw UDS)> send 0x2E 0x06 0x00 0x0A 0x19 0x00 0x32 0x24 0x06 0x01 0x0B 0x00 0x02
received 0x6E 0x06 0x00

```
Example #3 – Tester Present
`(raw UDS)> send 0x3E
`

bri3d, 2022-02-08:
Yes, this is a great idea - I have a similar tool I use myself, so this would mostly just be a matter of pasting it in, and then you get to take advantage of the large variety of backend interface support. 

I will get to this later this week. I will probably add macro commands to the REPL for some common tasks like clearing DTCs, sending a Workshop Code / Fingerprint, and setting Long Coding.

polydroi, 2022-03-07:
Hi, for such a use case I've extended scapy a while ago. Maybe you can make use of it: https://scapy.readthedocs.io/en/latest/layers/automotive.html#uds

bri3d, 2022-03-07:
This is pretty cool, I might try out the `scapy` functionality and see if I can add some plugins like Seed/Key over there if I like it.

polydroi, 2022-03-08:
I've implemented some basic security access as part of a UDS scanner, but the authentication function can be used stand alone aswell. Maybe you want to have a look at the following class, which should save some work, if you derive from it https://scapy.readthedocs.io/en/latest/api/scapy.contrib.automotive.uds_scan.html#scapy.contrib.automotive.uds_scan.UDS_SA_XOR_Enumerator.get_security_access