bri3d/VW_Flash #41: J2534 issue

Vadim88226, 2022-04-04:
Hello,

I have installed J2534 openport2.0 driver and run the GUI version(VW_Flash_GUI.py).
But it couldn't find the J2534 interface.
I have tried to modify the code in get_dlls_from_registry.
```
    try:
        BaseKey = winreg.OpenKeyEx(
            winreg.HKEY_LOCAL_MACHINE, r"Software\\WOW6432Node\\PassThruSupport.04.04\\"
        )
    except:
        logger.error("No J2534 DLLs found in HKLM PassThruSupport. Continuing anyway.")
        return interfaces
```
Then, script can find J2534 interface, but if I click the button "Get ECU info", this error occurred, OSError: [WinError 193] %1 is not a valid Win32 application.

bri3d, 2022-04-04:
You are using 64 bit Python with 32 bit drivers. Either install 32 bit
Python or 64 bit drivers. My recommendation is 32 bit Python as we have
seen mysterious issues with 64 bit OpenPort drivers.

Vadim88226, 2022-04-04:
Thanks for your response. I will try with 32 bit python version.
If I use 32 bit version, do I not need to modify the code in registry path?

bri3d, 2022-04-04:
That's correct, the code is written for use with 32 bit Python, 32 bit OpenPort drivers, on 64 bit Windows as this is what almost all users are running.

closed 2023-02-04 by bri3d