bri3d/VW_Flash #43: vas5054

R3m1ngt0n, 2022-04-12:
Can I use vas tool? How?

bri3d, 2022-06-15:
You should be able to just install the VAS5054 J2534 PassThru drivers and it will become a selectable interface. Any J2534 device should appear in the list and try to work. So far we have tested with OpenPort 2.0, OpenPort clones (some work and some don't), and several types and models of DrewTech cables (Mongoose etc.). 

It is not tested so it may or may not work - while J2534 is a standard, it is a very sloppy one, and not all hardware works the same way. 

If you try and it doesn't work, send the required logs (console output, flash.log, udsoncan.log, flash_details.log) and I can try to figure out why.

reikred, 2023-11-22:
Along the same lines, is there any knowledge as to which dongle chipsets supports J2534 in the right way, for example (or in particular) the STMIN_TX parameter and associated ioctl call? Maybe could start a thread where people could submit their experuiences as to which dongles work or not.

kamold, 2023-11-22:
Works perfectly with Scanmatik Pro2 clone in my experience.