bri3d/VW_Flash #44: Checksum issue with A7 Simos18.1 ECU

user1272349087234, 2022-04-27:
Hello,

there is an issue if you want to write only calibration maps after unlock with vwflasher. This issue is with stock file and modded file. Log File is attached. 
[flash_details.log](https://github.com/bri3d/VW_Flash/files/8573234/flash_details.log)
[flash.log](https://github.com/bri3d/VW_Flash/files/8573236/flash.log)

user1272349087234, 2022-04-27:
[full_bins.zip](https://github.com/bri3d/VW_Flash/files/8573244/full_bins.zip)
This both files are stock and not working...

bri3d, 2022-04-27:
https://github.com/bri3d/VW_Flash/blob/master/data/box_codes.csv has this file added to the database now. If you are using a Release instead of from Git directly, you can download this file, copy it into the data directory in your VW_Flash installation, and you should be good to go.