bri3d/VW_Flash #49: Question regarding bootloaders

issigg, 2022-05-14:
Hello! I am trying to unlock and flash a Simos 18.1 in a Skoda Octavia RS.
When ID:ing the ECU I get: 
`VW ECU Hardware Number : 06K907425B 
VW ECU Hardware Version Number : H13
VW Spare Part Number : 5G0906259  
VW Application Software Version Number : 0005
Boot Loader Identification : SC8.1 CB.00.00.80 C02.00 SC8 
VW System Name Or Engine Type : 2.0l R4 TFSI 
Engine Code Letters : CHHB`

In the windows.md doc, it says "If your Boot Loader Identification said SC8, you need FL_8V0906259H__0001.frf.
If your Boot Loader Identification said SCG, you need FL_5G0906259Q__0005.frf". 

However this ECU says it has the SC8 bootloader but different firmware.
Which one should I use for unlocking and flashing then?

Thanks lots for this!

bri3d, 2022-05-14:
Unlock using FL_8V0906259H__0001.frf. But do NOT flash after unlocking
using this FRF, use a matching one for your car like 5G0906259__0005. This
is how things are laid out in the documentation. Be very careful not to use
Full Flash Unlocked with that first 8V FRF as it will cause an immo lock.
Only use Unlock with the 8V FRF.

issigg, 2022-05-14:
Alright I got you. Didn't make that out of the docs sorry!
I guess flashing using a later version such as 5G0906259__0007 would be fine also?
Thanks for the answer will try it tonight.

bri3d, 2022-05-14:
Yep should be fine! 👍

issigg, 2022-05-15:
Worked great. Process didn't take long at all. Used an Openport 2.0 clone.

closed 2022-05-15 by issigg