bri3d/VW_Flash #52: Simos18.1 ecm disabled

VagArena, 2022-06-10:
Hello 
If i have simos18.1 for chhb engine with immo off 
When i tried instructions unlocked then flashed it wiped immo data i guess
What steps without touch immo data blocks to upload tune?
 because editing fullread write (iflash files) on bench as you know to get back immo and car run.

Thanks for your efforts

bri3d, 2022-06-10:
Immo Off on Simos18 is normally done in the Application Software by patching the immobilizer module's code rather than in EEPROM/DFlash as it has been done in some past ECUs.

Each time you full flash you need to flash a file with modified Application Software with Immo Off. If you flash a stock FRF or a BIN without an Immo Off applied, the ECU will re-lock.

If you don't want to flash patched ASW each time, use Flash Calibration to flash only the Calibration area.

VagArena, 2022-06-10:
So i did it with full unlocked flash and its not run, after i read same eeprom and wrote immo again in iflash then flashed and it back again as you said and it work.

Now 
i just flash use "unlocked calibration " only ?
Or start with unlock bootloader frf first then calibration?

About dflash patching could you help with info or guide alittle hint  

Sorry for many questions you are the only one knows if you want to msg you email write it to me please
Thanks in advance