bri3d/VW_Flash #56: Logging mode Issue

Vadim88226, 2022-06-16:
I tried to get data logging with mode 0x22, it seems it doesn't support 0x22 mode.
Also 3E mode doesn't work correctly, as I checked the csv file stored in local logging directory.
Please let me know how to solve these issue.

bri3d, 2022-06-16:
The logging system needs a YAML file provided with the correct data. The 3E system needs both an ECU patch and the correct RAM addresses from an A2L. I recommend you buy VehiCAL from http://vehical.net , the logging system is not really very complete and is quite challenging to set up. I am working on a new PC logger for $22 but it will be a few weeks at least before it is ready.

bri3d, 2022-10-26:
The logger has been almost completely rewritten and should be fixed.

closed 2022-10-26 by bri3d