bri3d/VW_Flash #57: issue with "Full Flash Unlocked (BIN/FRF)"

ButcherDSM, 2022-06-24:
Hello

after successful unlock, on level Full Flash Unlocked (BIN/FRF) - nothing happens and in logs i see some errors
Flash Stock (Re-Lock) also not work and car dont start now (

please help me, thanks!

all files in attac
[simos 18_1.zip](https://github.com/bri3d/VW_Flash/files/8974858/simos.18_1.zip)
h

sagishm, 2022-06-24:
Your ecu is locked and immo not passed, you need to read ecu in boot and so immo off, then you need to write original frf.

ButcherDSM, 2022-06-24:
so, my friend with Magic Flex make "restore to original" and car alive.
i cant use your VW_Flash for my car?

sagishm, 2022-06-24:
You can, follow the instructions step by step.

ButcherDSM, 2022-06-24:
i follow instructions step by step and got error
i do something wrong?

sagishm, 2022-06-24:
You selected the wrong unlock, you needs to select “Unlock ECU (FRF)” with the unlock file, then you needs to “Full flash unlocked” with your file.

bri3d, 2022-06-24:
> VW_flash starts unpacking something and stops at 80 with an error in the log window

Please post errors if you have them. If you don't post the error, I can't look into it...

ButcherDSM, 2022-06-24:
> You selected the wrong unlock, you needs to select “Unlock ECU (FRF)” with the unlock file, then you needs to “Full flash unlocked” with your file.

i do exactly that
first unlock goes fine
in the second step when i choose full flash unlocked - its freezes at 80

ButcherDSM, 2022-06-24:
> I do see in the logs that you posted (thank you for doing this! I didn't see them at first! I need to improve error logging for exceptions so that they show in logs too...) that something strange might be going on here.
> 
> The logs end with
> 
> `2022-06-24 10:59:47 [INFO] FlashUtils: Encrypting FD_1 compressed size :697620`
> 
> This suggests to me that there might be an issue with the padding being applied by `lzss` as this does not correctly end on a 16 byte boundary.
> 
> What version of VW_Flash are you using? Are you using a released version or a version you made yourself?
> 
> I have recently made changes to the compressor to support "exact length" control modules. These changes are not yet in a "stable" release (I try very hard to make sure stable releases are truly stable), but they are in pre-releases and Git. They should not have an effect on Simos compression, and I did test via both automation and flashing my own car to verify, but these files particularly may be producing a side effect I didn't notice. I'll take a look.

exactly! in error exist srting ".... not correctly end on a 16 byte boundary."
i use latest release, but not stable

i try stable version in few days and write to you
thanks for your help!

bri3d, 2022-06-24:
I fixed this on latest `master` . I also deleted the pre-releases as it seems they were causing issues. I will push another pre-release when I have some more new features ready to test.

Thank you for testing and reporting the issue!

closed 2022-06-24 by bri3d