bri3d/VW_Flash #61: UDS programming sessions available on the Bench??  "ConditionsNotCorrect" (0x22)

MBTune, 2022-07-24:
Have a 18.1 ECU on the Bench, Can ID etc etc but when trying to enter a programming session getting the following..

**_Server refused our request for service DiagnosticSessionControl with code "ConditionsNotCorrect" (0x22)_**

Would this be due to missing Gateway etc???

bri3d, 2022-07-24:
This is due to locked immobilizer. You need cluster and gateway or immo off
/ preconditions bypass patches in ASW.

MBTune, 2022-07-24:
Thank you, i guessed it would be IMMO/Gateway related, will see about doing ASW Patch if possible

bri3d, 2022-07-28:
Closing this out - 

to patch ASW to bypass IMMO, you also need a bench tool - since you can't flash ASW until immo is unlocked! I haven't found a way around this without opening the ECU.