bri3d/VW_Flash #63: Audi A5 Simos18.1 checksum error with "Calibration Flash Unlocked" writing

PwrMusic, 2022-08-04:
Hello! Thank you for your amazing work! 
I have a problem when I writing calibrations with "Calibration Flash Unlocked" on an Audi A5 Simos 18.1: "[ERROR] Checksum: Could not find ECM3 location for SC800H14". The "Full Flash Unlocked" writing without problems. On the Audi Q5 with the same block, but with a different version of software, the calibrations are writing without any problems. 
I am attaching the log files with 2 logs:  
2022-07-14 22:40:00 - unsuccessful "Calibration Flash Unlocked";  
2022-07-15 00:05:53 - successful "Full Flash Unlocked".

[flash.log](https://github.com/bri3d/VW_Flash/files/9255879/flash.log)
[flash_details.log](https://github.com/bri3d/VW_Flash/files/9255881/flash_details.log)
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/9255882/udsoncan.log)

bri3d, 2022-08-04:
Thanks for following the bug reporting guidelines!

This file wasn't in the boxcode database / CSV that the Calibration Flash feature uses to find the monitoring checksum areas. 

Full Flash will work because it looks into the application software code to find the monitored areas, instead of relying on a database. 

I'll look to add this software version (H14) to the box code database today. I thought I had finally gotten close to finding all of the Simos18 versions out there, but it looks like there are always more! :)

bri3d, 2022-08-04:
https://github.com/bri3d/VW_Flash/suites/7669804109/artifacts/319789209

This build should work. I will cut a new actual release in the next few weeks.