bri3d/VW_Flash #66: stupid operating error during flashing

en3rgy97, 2022-08-26:
Hello,

I made a mistake when unlocking the ecu.

I first did "Unlock ECU (FRF)" with the file "8V0906259H". The flashing was interrupted during the patching process, because of a loose contact. Then I restarted the vw_flash gui and clicked the wrong way and accidentally selected "Full FLash Unlocked (BIN/FRF)" and flashed the file "8V0906259H". I noticed this afterwards and wanted to flash the original frf from my car "5G0906259K" but VW-Flash got stuck at "SETUP - Upgrading to programming session.... - 100". I tried to flash with VCP to original but always get the error "programming preconditions not fulfilled :General reject 653".

What can I do to save the ECU ?

Translated with www.DeepL.com/Translator (free version)

bri3d, 2022-08-26:
It is immobilizer locked. You need to find someone with a bench flashing
tool.

sagishm, 2022-08-26:
There is bench tool for simos18?
I know there is only boot.

bri3d, 2022-08-26:
Whatever industry term you choose, the tool that lets you access flash
directly. I don’t know if any way without opening the ECU either.

Basically what happens is when you flash a file that does not have a
matching power class byte in the calibration, the immobilizer will never
let you flash this ECU again unless it is put in a car that matches. I
think it’s a bug and VW never really intended for this, but that’s the way
it is.

You need to use a tool which can write flash directly to write a file
matching the immobilizer power class data for the ECU to work again.

en3rgy97, 2022-08-26:
> It is immobilizer locked. You need to find someone with a bench flashing tool.
> […](#)

I have someone With Bench Tool. Did i can flash from extracted frf or did i need full read from someone Else? Sorry If offtopic i Just dont know who to ask about this.

sagishm, 2022-08-26:
Once it was happened to me i did immo off then i wrote frf in bench.
So basically the solution for that is to open ecu and read in boot mode, you can find someone to fix the IROM like @bri3d explained or disable the immo, then you can flash frf again.

bri3d, 2022-08-26:
Whether you can flash from extracted FRF depends on your bench tool software unfortunately. The extracted FRF -> BIN file does not contain the validity markers for the software blocks as they are outside the area which is port flashed, so they are not present in the FRF. It also does not contain the supplier bootloader. I have heard varying reports on how this works with commercial tools.

If you have an immo off solution, this way will work yes - just read, immo off, and write back. 

The other way to do this is to combine a bench read with the FRF->BIN data. Take a bench read using your tool, and in the regions which are 0 in the FRF-> BIN binary, copy the data from the bench read back into the converted FRF BIN. Most importantly, the area from 0 to 0x14000 (supplier bootloader) and the small areas containing the string H$B or B$H from after each software block (these are the Valid flags).

en3rgy97, 2022-08-29:
I have now a full Boot read.
Im not 100% Sure how to integrate  the extracted frf into my full read. 
Can maybe Someone Help merging this?

sagishm, 2022-08-29:
Just do immo off then write back, after that use obd and vw_flash to wrote original frf.

en3rgy97, 2022-09-01:
I got it Back With immo off Patch in Boot.
Thanks for the Help.

closed 2022-09-01 by en3rgy97