bri3d/VW_Flash #81: vw flash

revtekpeformance, 2022-11-22:
good day, im having a problem using the vw flash, i cant seem to get the unlock feature to work, i can ID the car perfectly but then cant get to flash functions for some reason , im on windows 11 using the latest GUI.
[flash_details.log](https://github.com/bri3d/VW_Flash/files/10069732/flash_details.log)
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/10069733/udsoncan.log)
[flash.log](https://github.com/bri3d/VW_Flash/files/10069734/flash.log)

bri3d, 2022-11-22:
What do you mean by "can't get to the flash functions?"

What kind of interface devices are you using? What I see in the logs is this:

You successfully Get Info using a J2534 driver with

```
J2534 FirmwareVersion: b'FW:0808 HW:02 SN:A55259', dllVersion: b'1.0.0.15 (built 12.02.19)', apiVersionb'04.04'
```

And also a J2534 driver with:

```
2022-11-22 12:33:29 [INFO] Connection: J2534 FirmwareVersion: b'1.17.4877', dllVersoin: b'1.02.4820 Jul  6 2016 17:20:04', apiVersionb'04.04'
```

But when you go to flash, you have a driver with

```
2022-11-22 19:26:59 [INFO] Connection: J2534 FirmwareVersion: b'', dllVersoin: b'', apiVersionb''
2022-11-22 19:26:59 [INFO] Connection: Failed to set ISO15765_STMIN to 0
2022-11-22 19:26:59 [INFO] Connection: Failure message: b'invalid channel ID'
2022-11-22 19:26:59 [INFO] Connection: Failed to set ISO15765_STMIN_TX: Error_ID.ERR_INVALID_CHANNEL_ID
```

- a driver which returns empty version strings and cannot handle the STMIN_TX ISO15765 command.

revtekpeformance, 2022-11-22:
,Ok so when i select unlock ecu nothing happens ,i dont have any 'flash' button on the interface ,so sorry im new to this just trying to flash my mk7 gti , im using scanmatik 2 clone and have a tatcrix clone aswell, i had thought my j2534 devices where the problem, if you recommend i can try with vas 5054

revtekpeformance, 2022-11-22:
[image]

bri3d, 2022-11-22:
Oh, I see! So weird - this is a purely GUI issue, not an issue with an interface device or anything else.

Maybe a weird issue with display scaling or the vertical resolution of your screen? I'd try messing around with display scaling and vertical screen resolution settings to see if you can get the button back.

Based on the logs whatever the last device you tried was isn't going to work, but the other two should be fine :)

revtekpeformance, 2022-11-22:
great thanks so much for the help and fast response much appreciated ill send an update once i resolve the display issue, thanks so much again :D

bri3d, 2022-11-22:
This appears to be a bug, but one in an error message.

It doesn't think your file matches the correct unlock file name. In the case of an SC8 ECU, you need `FL_8V0906259H__0001.frf` . It's failing to display the correct error message due to a bug which I will fix.

revtekpeformance, 2022-11-22:
ok great thanks its stuck on extracting file for unlock

mbperf, 2022-12-02:
Hi, I am using python 3.9 32-bit and it does not want to open vw flash gui.
Window open for 1 second and nothing happens afterwards.

Any suggestions? thx :)

zjmperformance, 2024-10-29:
> This appears to be a bug, but one in an error message.
> 
> 
> 
> It doesn't think your file matches the correct unlock file name. In the case of an SC8 ECU, you need `FL_8V0906259H__0001.frf` . It's failing to display the correct error message due to a bug which I will fix. 

Has this been fixed? I'm having same issue and now I have a car that doesn't move.

Rick-2022, 2025-01-26:
Would this software work with simos 3.3 

Kind regards