bri3d/VW_Flash #83: Few questions about unlock

averagejoe8, 2022-12-11:
Hi , 
First of all, many thanks for this great software. You've done a great job.

I want to ask few questions about unlock:
1) If ECU was previously tuned with other equipment like MMS FLEX and it is unlocked for tuning with this tool , can I write only calibration with Your software , without unlocking at the beginning ? Every tool does unlock Simos18.1 same way ?

2)Is there a way to check if ECU was unlocked before ? Flex for example is showing LOCKED or UNLOCKED when id is done.

3)My ecu is FL_5G0906259H_0001 info from Your box_codes file:
5G0906259H	1	2.0l R4 TFSI	SC8I0S00	SC8I0S50	SC800S50	55724	66096	['06K907425B', '5G0906259H']
So if i understand correctly i have to :
 1) Unlock it with FL_8V0906259H__0001.frf (unlock is always with this file regardless what is inside ecu?)
 2)Flash it with full .bin unpacked from : FL_5G0906259H_0001.frf

4) I have also a spare ecu that i want to immo off and use for development. If i understand correctly i have to :
1) read it in boot mode 
2)use vehical or other software to do immo off
3)unlock it with FL_8V0906259H__0001.frf
4)Flash it with full .bin unpacked from : FL_5G0906259H_0001.frf
Immo will stay off , or i would need to read ecu in boot mode and use immo off software again ?
Or is there other way to o it ?

ConnorHowell, 2022-12-12:
Regarding unlocking, yes since it's Simos 18.1 (SC8) you want "FL_8V0906259H__0001.frf". The only reason you'd need a different file is for Simos 18.10 (SCG). Once the unlock procedure is done you can then full flash the correct FRF/bin for the car. From that point onwards you can either do a unlocked full or calibration flash from within the GUI.

You won't need to redo the immo off as that's stored in DFLASH (EEPROM) and isn't written during any flashing process within VW_Flash. In case you don't already have a boot tool you can use the following made by Brian to build your own for this ECU: [bri3d/TC1791_CAN_BSL ](https://github.com/bri3d/TC1791_CAN_BSL)

bri3d, 2022-12-14:
Simos18 Immo Off is not done in DFlash alone. Some tools correct and re-encrypt the VIN in ImoDat in the DFlash, but all Immo Off files alter the ASW to hardcode the `stStatFact` to a specific value, bypassing the immo logic.

To answer your questions:

1. I have no idea what any commercial flashing tools do really. If they just put CBOOT into Sample Mode like my patches, then they will be compatible. If they have some proprietary protocol add-on or a different method of bypassing signature checking, they won't. Unfortunately I can't really guarantee anything when it comes to commercial tools as I have no idea what weird additional patches they may have applied.
2. Yes, but not right now. I should add this feature. Checking the Hardware Number _while the ECU is already in programming mode_ is the easiest tell. I will look into adding a button for this.
3. Yes, your understanding of the unlock process is correct.

As for Immo Off:

2)use vehical or other software to do immo off

Yes, and note that you need to use a bench/boot tool to write immo off the first time, you can't flash over UDS while Immo is active.

3)unlock it with FL_8V0906259H__0001.frf

Yes.

4)Flash it with full .bin unpacked from : FL_5G0906259H_0001.frf

No, just flash it with the .bin you already got back from your immo off service. This way you'll keep the immo off.