bri3d/VW_Flash #86: Unlock gone wrong

Roybee89, 2023-01-24:
[flash.log](https://github.com/bri3d/VW_Flash/files/10487115/flash.log)
[flash_details.log](https://github.com/bri3d/VW_Flash/files/10487116/flash_details.log)
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/10487118/udsoncan.log)

we are following the steps https://github.com/bri3d/VW_Flash/blob/master/docs/windows.md

we find ourselves stuck at  not seeing"H13" to  "X13"

We can no longer start the car or revert back to stock.

aarons3, 2023-01-24:
It looks like you did a normal full flash of the 8V0906259H file, NOT an "unlock" flash.  As a result, the ecu is now immo bricked and will need to be opened and recovered via bench.

Roybee89, 2023-01-24:
is there no way to recovery via VW_Flash?

aarons3, 2023-01-24:
Unfortunately not, I did the same thing to my own ECU once.

FirewinX, 2023-01-25:
in the log i see the following line "2023-01-23 20:24:47 [INFO] FlashUtils: Patching CBOOT into Sample Mode."
and the result is 8V0906259H this is correct.
but did you do this task in the "file" menu with "unlock ecu"?

bri3d, 2023-02-04:
Sorry this happened to you.

You _must_ unlock using File->Unlock ECU, or you risk an immobilizer brick.

You must be very careful to never flash a calibration with an ImoDat which does not match your car's PClass. IMO this is a bug in VW's immobilizer system but it cannot be bypassed besides by using boot flashing - once the PClass array in the calibration does not match the one stored in your immobilizer data, the car will not enter a programming session.

closed 2023-02-04 by bri3d