bri3d/VW_Flash #87: Logging Error

hybridr, 2023-01-26:
Every time I try to start a log, it shows the following errors:

Failed to create 3E list: 7f3e13
Unhandled exception: name 'exit' is not defined

[simos_hsl.log](https://github.com/bri3d/VW_Flash/files/10511788/simos_hsl.log)

Switchleg1, 2023-01-26:
This usually indicates your current ECU asw does not contain the SL HSL
patch.

hybridr, 2023-01-26:
Thanks for the answer. Can you please let me know what the patching process entails? Will this affect the current tuning on the car? I wasn't using VW_Flash for tuning. I just wanted to log with it. Thank you.

hybridr, 2023-01-26:
Much appreciated. I will try some things here and see if it works. Thank you.

hybridr, 2023-02-02:
> You would need to select mode 22 in this case. Mode 22 is available on all simos 18 ecus and is still very useful for logging your current tune. Modes 3E and HSL are faster and log key variables for tuning waste gate tables not available in mode 22 but require a patched bin. If wanted to use the HSL you could flash a freshly patched stock bin though you would not retain your current tune.
> […](#)

How can I patch a 8V0906259K full file to activate HSL? Please help. Thank you.