bri3d/VW_Flash #88: Wrong FRF flashed

nabsabs, 2023-03-15:
Hello,
I was trying to follow the instructions
My original ID showed Bootloader Identification as SCG
So I flashed the FL_5G0906259Q__0005.frf
And followed instructions and then flashed the correspondent bin file
Now the car does not start, I tried to flash back my stock file with HP tuners, this also did not work.

After investigating my issue and checking my original ID my original SW number was 8V0906259Q   0002
and now it is 5G0906259Q   0005

No device is being able to flash back the my original Bin

even trying to flash the new stock file is not working

attached are a few files and logs that I have

[ECU ID.txt](https://github.com/bri3d/VW_Flash/files/10978027/ECU.ID.txt)
[Successful unlock log.txt](https://github.com/bri3d/VW_Flash/files/10978028/Successful.unlock.log.txt)
[Trying to recover log.txt](https://github.com/bri3d/VW_Flash/files/10978032/Trying.to.recover.log.txt)

Magnum828, 2023-03-15:
It appears that you have immobricked your ECU. You were supposed to only use the 5G0 frf as an unlock FRF and then use extracted bin from FRF compatible with your 8V0 ECU. For now I think the only option is to construct bench equipment based on

nabsabs, 2023-03-15:
Hello,
I have found someone with a Flex device and I have provided the stock bin file (Size 6144 KB) based on my original ID
Do you think this file will restore the car to stock? because I was told that the coding is missing on my ecu and will require online update to fix

attached is my the screenshot that the technician sent me to show me that the coding is missing

[image]

Magnum828, 2023-03-15:
Coding is saved in the logs you provided so it should be as easy as copy and code through VCDS/ODIS/OBDeleven. Regarding Flex bench flashing and the file you provided I have no idea.

wiskiTS, 2023-03-16:
So based on this vcds id your ecu still got comms? Did you try to force update it with flex?

nabsabs, 2023-03-17:
Hello,
Its been a few days.
I took the car to a few technicians
They tried to do it online and in a few different solutions nothing is working

they are saying they cannot make any adaptation to the  immobilizer

The only way is to make an immo off, when they are doing so the car is starting but the problem is if I try to flash any file by obd (HP tuners or Kess) the car will not start and they will have to do the immo off again

can you help me further?

Is there a way that you can log in by teamviewer and try to make a fix ( I think people here have limited knowledge)

Also I am willing to pay for a correct fix that will restore my car to the conditions it was before

wiskiTS, 2023-03-17:
if you have tools to make  an boot/write  send them to me : info@tunningservice.com

jared52005, 2023-03-17:
Simple immo bypass which works on all ECUs and TCUs, including MDG1:
- Read the ECU in bench mode
- Delete (replace with 00) the whole calibration or ASW sector. One block is enough.
- Load this corrupted image back. Do not do CRC.

ECU during the booting process checks the integrity of ASW+CAL sector. If this check fails, ECU will stay in the bootloader, but immobilizer won't be enabled. You can then flash your desired boxcode. 

When you flash correct boxcode, which immobilizer is paired with, then it will release automatically.

bri3d, 2023-03-17:
> * Delete (replace with 00) the whole calibration or ASW sector. One block is enough.

Even one byte is enough, really, since all you're doing is forcing the ECU to stay in CBOOT. This is how I flash ECUs on the bench with my own tools, since the OEM flashing routines are faster than my own bootstrap loader - I erase a single flash region, then reboot into CBOOT and flash over UDS.

The issue I've had with recommending this approach to others with commercial tools is that most of them are crap and end up confused by the idea of flashing such a file - either they force-correct the checksums, refuse to write the zeroed region, or some other nonsense.

bri3d, 2023-03-17:
> The only way is to make an immo off, when they are doing so the car is starting but the problem is if I try to flash any file by obd (HP tuners or Kess) the car will not start and they will have to do the immo off again

You should not need an immo off file, just one with the tunIdx array matching the immobilizer data that is in your car (that is - a matching box code to what came on the car to start with).

Whatever file you're trying to flash over OBD must not have a matching tunIdx to the ImoDat in your car - are you sure it's a correct file matching the original box code? Hopefully trying to adapt the ECU didn't mess something up.

closed 2023-12-31 by bri3d