bri3d/VW_Flash #91: gaps between VW and m74.9

rusefillc, 2023-04-25:
Hello Brian

We have two UDS firmware update traces at https://github.com/rusefi/m74.9/tree/main/firmware/oem see .xls files

https://github.com/rusefi/m74.9/blob/main/firmware/oem/FirmwareUpdate.md has some UDS dialect notes and even secrets

Question: any chance you could glance and evaluate the gaps between this VW_Flash and being able to flash m74.9?

The end results is being able to flash rusEFI into OEM m74.9 hardware without opening the case

bri3d, 2023-04-25:
Hi!

Using a cheap OEM ECU to run rusEFI is very cool! Is this really an ECU powered by a knockoff Chinese STM32? That's amazing, I would never have expected to see such a thing.

Why not just write a simple script using the `udsoncan` module in Python which will perform these flashing steps? It should be just a few lines of code, really.

The basic `used for uploading main firmware` steps could on paper be replicated by VW_Flash with some tweaking but it would be a pretty significant departure and it does not look like you would take advantage of any of the VW specific infrastructure (LZSS, SA2 Seed/Key, etc.) which VW_Flash has. I don't see a real reason to use VW_Flash and not make a new tool instead.

The weird memory bootloader would be a complete departure and needs its own tool.

Do you know the checksum algorithm / range? That seems like it must be the main thing you're missing to be able to flash, unless there are also internal signatures or checksums inside the uploaded binary. Have you successfully written a modified firmware yet?

rusefillc, 2023-04-25:
@bri3d this world has changed in Feb of 2022. https://github.com/rusefi/m74.9 and https://github.com/rusefi/s105 are two consequence of those events.

Thank you for the https://udsoncan.readthedocs.io/en/latest/ hint!

bri3d, 2023-04-26:
This is extremely interesting to learn - it makes sense, of course, politically. But - I'm just curious, was it STM32 ARM before the supply chain changed, and Artery now? Even that would be surprising to me, STM32 is not a safety/ASIL approved processor so it is almost never seen in automotive applications (besides rusEFI, of course!!!). 

Yes, I'd just use udsoncan and start from scratch - not that I don't like you, I just think it will be easier to start from scratch with a purpose made tool as there is not much there. The interesting parts of VW_Flash were the compression, SA2, encryption, and signature bypass exploit, none of which apply to you :) 

Please let me know if you have any other questions or road blocks, this is a very interesting project to me.

rusefillc, 2023-04-26:
Two different stories
* s105: unsolder stm32, solder geehy, it _just_ works
* m74.9: 144 package has a couple of pins incompatible. We believe that m74.9 PCB is universal and it would accept either stm32 or artery. In terms of firmware still learning more, looks like there is major resemblance but again incompatible difference.

rusefillc, 2023-04-26:
> Please let me know if you have any other questions or road blocks, this is a very interesting project to me.

Thank you! At the moment still too few of those m74.9 in our possession, we are still mostly preparing for when we have more pieces to play with.