bri3d/VW_Flash #95: Issue while flashing, got stuck

abdmrknjc, 2023-05-05:
Hello, I have an issue regarding flashing, was flashing full flash unlocked method BIN file extracted from frf, and it got stuck at the end, SETUP finalizing, now I dont have comms on VCDS, no response from ecu, tried unlocking it again, first time unlock went fine, flashed normal frf, car started etc...
Help please
[flash.log](https://github.com/bri3d/VW_Flash/files/11408545/flash.log)
[flash_details.log](https://github.com/bri3d/VW_Flash/files/11408546/flash_details.log)
[udsoncan.zip](https://github.com/bri3d/VW_Flash/files/11408553/udsoncan.zip)

bri3d, 2023-05-05:
The last file you flashed looks like it had some kind of modification made
in the application software area based on the logs, I think you may have a
real brick. Did you try to apply some kind of software patch to this BIN?

abdmrknjc, 2023-05-05:
I checked file in WinOLS editor, provider edited more than calibration area, it was just supposed to be stage 1 tune, but I saw it, tried getting back to re-lock etc, but no success, it just stays there at entering diagnostic session after compressing FRF file

FirewinX, 2023-05-07:
give it a try to flash with odis. when the flash session starts you make a ignition change. i have with this a ecu recovered but a DCM 6.2V