bri3d/VW_Flash #96: Not an issue, precisions about DSG software update

cqke, 2023-05-22:
Hello,
Thank you for your tool. It's truely amazing.
I have a simple question for you : I want to flash a new tune for my DSG DQ250MQB.
Hardware Part Number is 0D9300020D, Revision 4308.
I only found an FRF called FL_0D9300020D_4310_OlCD_sw.frf.
If I extract the .bin, can I flash it with your tool without problems ?
I have a doubt with the VW Application Software Version Number, which is different.
If yes, does my car need a relearning of the TCU or something else ?

Thank you very much for your help.

ConnorHowell, 2023-05-23:
You say the hardware part number is 0D9300020D, the box code is **"VW Spare Part Number"** not the hardware number..

If your "VW Spare Part Number" is 0D9300020D and the "VW Application Software Version Number" is 4308 then you are fine to flash that FRF, the only difference between the 2 will be some minor calibration changes I suspect. It's just the latest sw revision for that code, if you took the car into the dealership for a software update that's the one they'd apply.

cqke, 2023-05-23:
Thanks you very much for your answer.

closed 2023-05-23 by cqke