bri3d/VW_Flash #97: Not issue

angelll690, 2023-05-30:
Hello 
One question:
This program calculate checksum after turning .bin?
Thanks in advance 
Thanks developers

ConnorHowell, 2023-06-06:
Yes, checksumming is implemented for: Simos18, DQ250 & DQ381.

angelll690, 2023-06-06:
Hello. I used your program 0.6.0 and everything worked like a charm. I
order tuned file and when I flash it(calibration) performance of my car not
changed, then I flash with (full flash) again nothing changed. What I’m
doing wrong?
Thanks
[image: Архів ZIP]
file-service-4098-modified.zi…
<https://drive.google.com/file/d/1W0PjirbESO7_u5VXOuw5uRs7Y2-9M9oR/view?usp=drivesdk>
[image: Архів ZIP]
file-service-4098-original.zi…
<https://drive.google.com/file/d/1D4oqfMh4aXvWG9uyqet2H4Hj_GuqNN_3/view?usp=drivesdk>

ConnorHowell, 2023-06-06:
If you flashed that file and the vehicle started afterwards; it's an issue we can't resolve and it's 100% down the tune produced by the file service. VW Flash wouldn't have altered anything in the supplied binary. 

A checksum issue would result in the vehicle not starting, regardless it looks like the file has been pre checksummed in WinOLS anyways, so this is almost certainly a tune issue.