bri3d/VW_Flash #103: Get ecu info not working (no connection)

DEN3535, 2023-07-16:
Hi!
Clone equipment based on ESP32S CP2102 TYPE-S
Everything works fine with Simontools.
Tiguan MQB 2.0 gen3 CHB unit 5N0906259 / ECU HWN 06K907425B
VW_Flash can't work. 
What could be the problem?
Running VW_Flash from Wine 10 x86
The ESP32 chip was flashing with different firmware!
So far nothing helps to establish a connection with the ECU!
I opened the hood of the car, washed the headlights, pumped up the wheels)

[flash.log](https://github.com/bri3d/VW_Flash/files/12064926/flash.log)
[flash_details.log](https://github.com/bri3d/VW_Flash/files/12064927/flash_details.log)
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/12064929/udsoncan.log)

DEN3535, 2023-07-20:
Please help me. Communication with the ecu is working! 
What file do i need to unlock ecu?
[IMG_20230719_133956_edit_3964411527556526](https://github.com/bri3d/VW_Flash/assets/139713302/1fa5ea98-23c9-4e44-98c4-0d28add85810)
[image]
[image]

DEN3535, 2023-07-20:
[image]

DEN3535, 2023-07-20:
! 
[image]

DEN3535, 2023-07-20:
Foto
[image]
[image]
!

qubex22, 2023-07-25:
> Hi! Clone equipment based on ESP32S CP2102 TYPE-S Everything works fine with Simontools. Tiguan MQB 2.0 gen3 CHHB unit 5N0906259 / ECU HWN 06K907425B VW_Flash can't work. What could be the problem? Running VW_Flash from Wine 10 x86 The ESP32 chip was flashing with different firmware! So far nothing helps to establish a connection with the ECU! I opened the hood of the car, washed the headlights, pumped up the wheels)
> 
> [flash.log](https://github.com/bri3d/VW_Flash/files/12064926/flash.log) [flash_details.log](https://github.com/bri3d/VW_Flash/files/12064927/flash_details.log) [udsoncan.log](https://github.com/bri3d/VW_Flash/files/12064929/udsoncan.log)

Hi, I have the clone based on esp32s also, but get Ecu info doesn't even work. How did you manage to make it work? which firmware did you flash?

Thanks

DEN3535, 2023-07-28:
Driver 0.9.  The laptop only worked on windows 10 x64!