bri3d/VW_Flash #106: Can't Get ECU Info

aws4273, 2023-09-08:
I've been able to connect to Macchina A0 and reflash it using the GUI. However, I can not seem to connect to the car. Get ECU Info gives the attached errors. Connected via USB.  2019 GTI, SIMOS 18.10.
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/12554903/udsoncan.log)

bri3d, 2023-09-08:
Something in your A0 firmware is crashing with an `abort` message. 

Instead of trying to go down the rabbit hole of figuring out what's broken, I recommend you just download the latest Switchleg1 J2534 A0 firmware from https://github.com/Switchleg1/esp32-isotp-ble-bridge/releases/tag/v0.90 , install it using the tool that comes with it, and switch to using this firmware over J2534.

I am planning to drop direct connect support for _any_ A0 firmware as they have all gone their own ways as separate projects, and to just support J2534 as a common interface going forward (with some hacks to do faux-J2534 on OSX and Linux).

aws4273, 2023-09-21:
Thanks, that worked!

closed 2023-12-31 by bri3d