bri3d/VW_Flash #108: Decode and decompress type AA (Audi)

vlbosch, 2023-09-21:
Hi!

While researching the Audi E-tron, I discovered that Audi reused the 95kWh battery from the E-tron 55 in the facelift Q8 E-tron 50. They also upped the usable kWh from 86,5kWh to 89kWh. They did something similar with the 2019 models, which got an upgrade from 83,6kWh to 86,5kWh. Therefore, I thought it should be possible to cross flash the firmware from the FL to the pre-FL. Unfortunately the new firmware isn't available to download and the HW-partnumber changed from 4KE915233B to 4Ke915233C. What I do have, is the original firmware-file with 83,6kWh and the update to 86,5kWh. Now, I would like to investigate what Audi changed to allow the netto capacity-increase. Maybe even dump the firmware from the FL and verify my findings.

I have a separate battery control module (J840) on the bench to test the changes. With VCP I got the ODX files from the FRF-files. Now I need to decode and decompress them. From the ODX I get the following information:
<ENCRYPT-COMPRESS-METHOD TYPE="A_BYTEFIELD">AA</ENCRYPT-COMPRESS-METHOD> I attached the two ODX files.

I found a post on Nefarious about extracting the BIN file and the Python-code to do so. Unfortunately it doesn't seem to work and it references the SIMOS18. Now this ECU isn't a motor control unit, so I figure that's the problem. After opening the bench module, I found the processor is a SPC5746CMKU6 1N06M, which is a NXP MPC5746C. 

Does anyone now how to extract the BIN files for this chip/controller? 

Thanks!
[VCP ODX.zip](https://github.com/bri3d/VW_Flash/files/12683857/VCP.ODX.zip)

ConnorHowell, 2023-09-23:
So although both Simos 18 and that ODX you have specify encryption/compression AA there's still every chance the underlying methods are different. More often than not it (AA) is referencing a combination of LZSS for compression and AES for encryption; but the AES key & IV will differ between control units, this is why it won't work for the ODX you've provided.

What you'd need is a dump of the flash from the device in order to disassemble and figure out the exact encryption/compression needed.

bri3d, 2024-09-25:
Closing this issue as Connor answered the question.

closed 2024-09-25 by bri3d