bri3d/VW_Flash #109: Program stops at "unlock ecu" with frf file

316tiSleeper, 2023-10-05:
I was trying the windows gui version for the first time...

I followed the instruction, found the compatible frf file for my car and tried to start unlock ecu -> select "FL_5G0906259N__0003_10375561AA.frf" -> unlock ecu

Some error messages show up in the console and the gui console stops at "extracting FRF for unlock..."

[flash.log](https://github.com/bri3d/VW_Flash/files/12821218/flash.log)
[flash_details.log](https://github.com/bri3d/VW_Flash/files/12821220/flash_details.log)
[udsoncan.log](https://github.co
[image]
m/bri3d/VW_Flash/files/12821222/udsoncan.log)
[image]

bri3d, 2023-10-05:
You have selected an FRF that is not the correct one for unlocking. There
is a bug where this displays a Python error instead of the correct help
message, but the root cause is using the wrong FRF

316tiSleeper, 2023-10-05:
Oh I'm sorry now i get it.
You always use FL_8V0906259H__0001.frf or FL_5G0906259Q__0005.frf for unlocking even if it doesnt match with your VW Spare Number at all.
From there forward you use your own matching FL_XX_XX.frf

Thank you!

316tiSleeper, 2023-10-06:
How can you see/test if a Simos 18 Car ist already unlocked?

closed 2023-12-31 by bri3d