bri3d/VW_Flash #112: Flash non-ECU hardware on the benchtop?

reikred, 2023-11-10:
Can VW_Flash be used to flash non-ECU hardware on the benchtop? In particular I want to flash the file FL_5C6915182___0601.frf into the 5C6915182B battery control module of a 2013 hybrid vw, via CANbus.

Basically what I am trying to do is to replicate the setup shown at the beginning of this video, but without spending $700 on a commercial tool and subscriptions.

https://www.youtube.com/watch?v=ti-m3lo53O4

The plan is to attempt benchtop flash operation (I think), because the BCM CANbus is no comm as seen from the obd2 connector, but I *could* also try via obd2 connector (incoming CANbus signals to BCM are clean, the unit just does not talk back, but that may be just because the BCM is bricked, which is what I am trying to fix in the first place).

If the above sounds plausible, is Tactrix OpenPort 2.0 J2534  a good choice for talking to CANbus directly?`

bri3d, 2023-11-11:
Flashing non-ECU hardware:

* Yes if it's UDS, but you need to write a `module` right now in `lib/modules` containing the SA2 script and description of flash layout first. This shouldn't be necessary long term and @ConnorHowell was working on direct/verbatim FRF flashing, but hasn't finished yet.

2013 is also a bit old for UDS - does this module talk UDS or TP2.0? If TP2.0, then no support yet.

However, there's another wrench: if your module is not communicating over OBD / CAN, then you probably shouldn't bother - reflashing over UDS is probably not possible; after all, UDS reflashing requires quite strictly that the module be talking over UDS. In this case you need a lower-level recovery - if it has a supplier bootloader, going in through that, or opening the module and reflashing it via hardware (direct chip off flash and/or some kind of bootstrap loader).

I can't tell what the commercial tool in that video is doing directly, but based on the logging output it seems (???) to be going over the diagnostic service and it's unclear to me that it could recover a brick which is not responding, unless it's using some custom backdoor "knock" implemented in the module.

reikred, 2023-11-13:
I'll add a little more information that may be relevant as to whether VW_flash might be able to write to my BCM module: 

Here is a picture of the flash interface internally on the board, with pins marked (GND, TXD, RXD, RST, PDX, VCC) from top to bottom. Is VW_Flash capable of talking to this interface, or are there other tools I should be considering?

[https://imgur.com/a/EE0SsgZ](url)

bri3d, 2023-11-13:
Out of scope. VW_Flash is for flashing VW control modules over UDS. We might add TP2.0 or KWP support in the far future. But direct hardware programming interfaces are device specific and demand device specific code (in many cases, just a generic programming tool for the MCU used on the device will work, depending on if it has protection enabled).

reikred, 2023-11-20:
I have found out that the CANbus is alive after all. By disconnecting BCM pin F8 from Vdd=12V, the BCM sends out about 4 seconds of CANbus signals, followed by 4 seconds of constant voltage, before going to 0 volts again. Repeated connects of F8 to Vdd then float yields the same behavior. 

 So, there is a hope that VW_flash can do the job, but I would have to learn some stuff about SA2 scripts and figure out the flash layout. I'll report back if any progress made, in case might be useful for other people.

reikred, 2023-11-24:
My dongle is arriving today, so now I am trying to learn how to write a module file. I am planning to call it lib/modules/FL_5C6915182___0601.py.

Q1: To generate the module file, is all the info needed included in the XML metadata that is part of the  FL_5C6915182___0601.odx file? 

Q2: It looks to me like trying to mimic dq250mqb.py is a reasonable starting point for FL_5C6915182___0601.py.  The problem is  to  understand how the XML values translate into the module file. But I can't find an .frf file for dq250 (to see how the parameter values in dq250mqb.py were derived). All I can find is a file called KD_DQ250_C3_05_002_sw.sgo, a binary file. If I knew how to make an .odx from from an .sgo I might be able to do something. Am I going in the right direction here? Did I overlook some documentation that explains how to generate the a lib/modules file?

For reference, here is the metadata from the .odx file:

[FL_5C6915182___0601.odx.metadata.txt](https://github.com/bri3d/VW_Flash/files/13461881/FL_5C6915182___0601.odx.metadata.txt)

reikred, 2023-11-25:
Here's the metadata in explicit form:

```
?xml version="1.0" encoding="UTF-8"?><ODX xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="odx.xsd" MODEL-VERSION="2.0.1">
  <FLASH ID="FL_5C69151820601">
    <SHORT-NAME>FL_5C69151820601</SHORT-NAME>
    <LONG-NAME>5C6 915 182__ 0601</LONG-NAME>
    <DESC>
      <p>ODX FLASH-Container beinhaltet alle Informationen zum Flashen eines Steuergeraets</p>
    </DESC>
    <ADMIN-DATA>
      <DOC-REVISIONS>
        <DOC-REVISION>
          
          <REVISION-LABEL>0601</REVISION-LABEL>
          <STATE>NEW</STATE>
          <DATE>2012-11-12T10:53:57</DATE>
          <TOOL>EB tresos Flashbuilder V2.1.5</TOOL>
          
        </DOC-REVISION>
      </DOC-REVISIONS>
    </ADMIN-DATA>
    
    <ECU-MEMS>
      <ECU-MEM ID="EMEM_5C69151820601">
        <SHORT-NAME>EMEM_5C69151820601</SHORT-NAME>
        <LONG-NAME>5C6 915 182__ 0601</LONG-NAME>
        <DESC>
          <p>ODX FLASH-Container beinhaltet alle Informationen zum Flashen eines Steuergeraets</p>
        </DESC>
        <MEM>
          <SESSIONS>
            <SESSION ID="EMEM_5C69151820601.SES_5C69151820601">
              <SHORT-NAME>SES_5C69151820601</SHORT-NAME>
              <LONG-NAME>5C6 915 182__ 0601</LONG-NAME>
              <EXPECTED-IDENTS>
                <EXPECTED-IDENT ID="EMEM_5C69151820601.SES_5C69151820601.EI_VWSparePartNumbe">
                  <SHORT-NAME>EI_VWSparePartNumbe</SHORT-NAME>
                  <LONG-NAME>VW Spare Part Number</LONG-NAME>
                  <IDENT-VALUES>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">8R0915591  </IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">5C6915182  </IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">5C6915182A </IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">5C6915182B </IDENT-VALUE>
                  </IDENT-VALUES>
                </EXPECTED-IDENT>
                <EXPECTED-IDENT ID="EMEM_5C69151820601.SES_5C69151820601.EI_VWECUHardwVersiNumbe">
                  <SHORT-NAME>EI_VWECUHardwVersiNumbe</SHORT-NAME>
                  <LONG-NAME>VW ECU Hardware Version Number</LONG-NAME>
                  <IDENT-VALUES>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">H03</IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">H04</IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">H05</IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">H06</IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">H07</IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">H08</IDENT-VALUE>
                    <IDENT-VALUE TYPE="A_ASCIISTRING">H09</IDENT-VALUE>
                  </IDENT-VALUES>
                </EXPECTED-IDENT>
              </EXPECTED-IDENTS>
              <SECURITYS>
                <SECURITY>
                  <SECURITY-METHOD TYPE="A_ASCIISTRING">SA2</SECURITY-METHOD>
                  <FW-SIGNATURE TYPE="A_BYTEFIELD">6805824A10680284921019734A05872506200382499389111975824A058712082001824A0181494C</FW-SIGNATURE>
                </SECURITY>
                <SECURITY>
                  <SECURITY-METHOD TYPE="A_ASCIISTRING">ALFID</SECURITY-METHOD>
                  <FW-SIGNATURE TYPE="A_BYTEFIELD">014101</FW-SIGNATURE>
                </SECURITY>
                <SECURITY>
[... 230 lines cut ...]
            <IDENT-IF-SNREF SHORT-NAME="EI_VWSparePartNumbe"/>
            <OUT-PARAM-IF-SNREF SHORT-NAME="Param_DataRecor"/>
          </IDENT-DESC>
          <IDENT-DESC>
            <DIAG-COMM-SNREF SHORT-NAME="DiagnServi_ReadDataByIdentVWECUHardwVersiNumbe"/>
            <IDENT-IF-SNREF SHORT-NAME="EI_VWECUHardwVersiNumbe"/>
            <OUT-PARAM-IF-SNREF SHORT-NAME="Param_DataRecor"/>
          </IDENT-DESC>
        </IDENT-DESCS>
        <ECU-MEM-REF ID-REF="EMEM_5C69151820601"/>
        <LAYER-REFS>
          <LAYER-REF DOCREF="EV_BECM0_S02" DOCTYPE="LAYER" ID-REF="EV_BECM0_S02"/>
          <LAYER-REF DOCREF="BV_BatteEnergContrModulUDS" DOCTYPE="LAYER" ID-REF="BV_BatteEnergContrModulUDS"/>
          <LAYER-REF DOCREF="EV_BECM0_002" DOCTYPE="LAYER" ID-REF="EV_BECM0_002"/>
          <LAYER-REF DOCREF="EV_BECM0_003" DOCTYPE="LAYER" ID-REF="EV_BECM0_003"/>
          <LAYER-REF DOCREF="EV_BECM06020115C6915182_004" DOCTYPE="LAYER" ID-REF="EV_BECM06020115C6915182_004"/>
          <LAYER-REF DOCREF="EV_BECM06020115C6915182A_005" DOCTYPE="LAYER" ID-REF="EV_BECM06020115C6915182A_005"/>
          <LAYER-REF DOCREF="EV_BECM06020115C6915182B_005" DOCTYPE="LAYER" ID-REF="EV_BECM06020115C6915182B_005"/>
          <LAYER-REF DOCREF="EV_BECM06020115C6915182_005" DOCTYPE="LAYER" ID-REF="EV_BECM06020115C6915182_005"/>
        </LAYER-REFS>
      </ECU-MEM-CONNECTOR>
    </ECU-MEM-CONNECTORS>
  </FLASH>
</ODX>
```