bri3d/VW_Flash #113: Error decrypting and unpacking .FRF file

reikred, 2023-11-22:
While waiting for a dongle to arrive, I am trying to decrypt and unpack an .frf file into a .bin file, but I fail. 

Any thoughts on what might be going wrong here? I'm on ubuntu 22.04.3, but the process was also not working in win10 (it was hanging in the GUI, I did not try the cmdline in windows).

```
python3 VW_Flash.py --action prepare --frf ../FL_5C6915182___0601.frf --output_bin ../FL_5C6915182___0601.bin
cannot import name 'WINFUNCTYPE' from 'ctypes' (/usr/lib/python3.10/ctypes/__init__.py)
dashing module not loaded
2023-11-22 11:43:10 [INFO] VWFlash: Starting VW_Flash.py
Traceback (most recent call last):
  File "/tmp/VW_Flash-0.6.0/VW_Flash.py", line 288, in <module>
    input_blocks = input_blocks_from_frf(args.frf)
  File "/tmp/VW_Flash-0.6.0/VW_Flash.py", line 269, in input_blocks_from_frf
    input_blocks[filename] = BlockData(i, flash_data[filename])
KeyError: 'FD_0'
```

bri3d, 2023-11-22:
The `prepare` feature is intended only for supported binaries: Simos ECU, DQ250, DQ381, and Haldex files, as it also performs checksumming and other operations.

The file `FL_5C6915182___0601.frf` is not one of these supported control modules (it's a BMS I guess?), so it crashes.

You can still use the tools in this repo to extract this file, though:

Use `PYTHONPATH=. python3 frf/extractfrf.py --file ../FL_5C6915182___0601.frf` to produce the ODX and then `python3 extractodx.py --file FL_5C6915182___0601.odx --outdir .` to produce the ODX segments.

You're in luck as this file doesn't look to be encrypted or compressed.

reikred, 2023-11-22:
Yeah, it is the BCM/BMS file. Does the above mean I  will not be able to load this .frf file into the BCM using VW_Flash? I did manage to create the .odx file, but is hat good enough?

reikred, 2023-11-23:
Just a small correction. I think you meant decryptfrf.py not extractfrf.py in the command

```
#PYTHONPATH=. python3 frf/extractfrf.py --file ../FL_5C6915182___0601.frf 
PYTHONPATH=. python3 frf/decryptfrf.py --file ../FL_5C6915182___0601.frf 
```

reikred, 2023-11-24:
Fro reference, here are the resulting image files for the whole process

```
% ls -l FD_*
-rw-r--r-- 1 reik reik   8192 Nov 23 13:51 FD_0DRIVE
-rw-r--r-- 1 reik reik 385024 Nov 23 13:51 FD_1DATA
-rw-r--r-- 1 reik reik   8192 Nov 23 13:51 FD_2DRIVE
-rw-r--r-- 1 reik reik   4096 Nov 23 13:51 FD_3DATA

```

reikred, 2023-11-30:
But that is exactly the problem: I don't know how to generate a binary flash file that places these 4 code+data segments at the correct locations  of the 512kB flash memory, where the boot code expects to find them, and without overwriting the boot code itself.

Am I explaining the problem the wrong way? 

I tried to figure out where the 4  segments are supposed to be located by hooking up an Orange5 eeprom programmer and trying to read out the current flash content. Then I could compare it to the segments extracted from the .frf/odx and find the addresses using some spatial correlation/matching of the data. BUT the big but is that the uPD76F0192 MCU chip is write-only erase-only and I cannot read out the old content. So I have nothing. Am I making sense?

reikred, 2023-12-01:
That's an interesting idea. Additional thought along those lines: For the code segments maybe running the bytes through an ARM disassembler could narrow down the possibilities of what bytes might be an address?

Update on what I have been up to: I tried to make an inversion/deduction attempt using crchack to find out what address value may have been prepadded or postpadded (cf.  "address-referenced checksum") to each segment that would result in the known (from metadata) CRC32 values. But it failed: It looks like the CRC32 for all segments were calculated using the same prepadded fake address of 0x0001000 hex = 64k start address. I know the address should be at most 19b  (512kB flash) of nonzero values, so could also determine that flipping the bits and also the postpad inversion gave no sensible addresses at all.

reikred, 2023-12-01:
Scavenging for possible addresses: Hmm, I cannot see anything obvious. The value 0001 5004 id within the 7FFFF range but it is the same in two of the files, so that seems like an unlikely candidate unless the 2n fle is just a backup copy intended to be loaded into the 1st destination 15004 if needed.

Any ideas? I could upload the files somewhere if that would help.

```
machine(user):6172 > hexdump -n 64 FL_5C6915182___0820/FD_0DRIVE
0000000 0001 5004 c014 ffff c054 ffff c076 ffff
0000010 c2aa ffff 0e40 0000 3761 cae1 6201 6746
0000020 0002 0746 0001 0746 0000 0746 0003 0766
0000030 0004 0766 0009 0766 000d 0766 0011 0766
0000040
machine(user):6173 > hexdump -n 64 FL_5C6915182___0820/FD_1DATA
0000000 0783 5810 0000 0000 0000 0000 0000 0000
0000010 0785 a4c8 0000 0000 0000 0000 0000 0000
0000020 0785 a4b8 0000 0000 0000 0000 0000 0000
0000030 0000 0000 0000 0000 0000 0000 0000 0000
0000040
machine(user):6174 > hexdump -n 64 FL_5C6915182___0820/FD_2DRIVE
0000000 0001 5004 c014 ffff c054 ffff c076 ffff
0000010 c2aa ffff 0e40 0000 3761 cae1 6201 6746
0000020 0002 0746 0001 0746 0000 0746 0003 0766
0000030 0004 0766 0009 0766 000d 0766 0011 0766
0000040
machine(user):6175 > hexdump -n 64 FL_5C6915182___0820/FD_3DATA
0000000 f398 fec2 3044 3130 0000 0000 000a 0000
0000010 0001 0000 0002 0032 0001 0000 e630 0002
0000020 0064 0000 0258 0000 0032 0000 028a 0000
0000030 00a0 0000 fed4 ffff 000a 0000 fed4 ffff
```

reikred, 2023-12-02:
I managed to get the ghidra tool going, and here is the result of disassembly of the FD_0DRIVE segment file. The analysis creates both assembly code and some C-like pseudo code, as enclosed below. Is there any address info embedded  here that indicates the starting address?

[FD_0DRIVE.c.txt](https://github.com/bri3d/VW_Flash/files/13533440/FD_0DRIVE.c.txt)

reikred, 2023-12-02:
I used Ghidra to create a full set of disassembled and decompiled files. Here they are:

[FD_0DRIVE.asm.txt](https://github.com/bri3d/VW_Flash/files/13536452/FD_0DRIVE.asm.txt)
[FD_0DRIVE.c.txt](https://github.com/bri3d/VW_Flash/files/13536453/FD_0DRIVE.c.txt)
[FD_1DATA.asm.txt](https://github.com/bri3d/VW_Flash/files/13536454/FD_1DATA.asm.txt)
[FD_1DATA.c.txt](https://github.com/bri3d/VW_Flash/files/13536456/FD_1DATA.c.txt)
[FD_2DRIVE.asm.txt](https://github.com/bri3d/VW_Flash/files/13536457/FD_2DRIVE.asm.txt)
[FD_2DRIVE.c.txt](https://github.com/bri3d/VW_Flash/files/13536458/FD_2DRIVE.c.txt)
[FD_3DATA.asm.txt](https://github.com/bri3d/VW_Flash/files/13536459/FD_3DATA.asm.txt)
[FD_3DATA.c.txt](https://github.com/bri3d/VW_Flash/files/13536460/FD_3DATA.c.txt)

Hopefully there is an expert that can help me deduce the correct flash memory address for each one of these ...