bri3d/VW_Flash #115: reflashing ao fail

bb-tt, 2023-12-12:
was following the vw flash gui guide and was doing the ao reflash last thing it said was something about hard reset then light on dongle went off waited about 10 minutes nothing. tried restarting everything now the power light does not come on on the ao dongle and the computer wont connect to it
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/13642285/udsoncan.log)
[flash_details.log](https://github.com/bri3d/VW_Flash/files/13642287/flash_details.log)

aarons3, 2023-12-12:
Try this:  https://docs.google.com/document/d/1owDMM_Z_GIP6NYBgR-IbAAWMnZRcDOJvEh1ru3ht_TQ

https://github.com/Switchleg1/esp32-isotp-ble-bridge/releases

bb-tt, 2023-12-12:
that worked thanks you can close this issue.

closed 2023-12-12 by bb-tt