bri3d/VW_Flash #116: Unlock SCG - problem

jasmuh1, 2023-12-20:
File / Unlock ECU
FL_5G0906259Q__0005.frf - **SCG**

**CMD:**
Traceback (most recent call last):
  File "VW_Flash_GUI.py", line 871, in on_select_unlock
  File "VW_Flash_GUI.py", line 384, in flash_unlock
  File "lib\extract_flash.py", line 12, in extract_flash_from_frf
  File "lib\extract_flash.py", line 28, in extract_data_from_odx
  File "extractodx.py", line 116, in extract_odx
  File "extractodx.py", line 68, in decompress_raw_lzss10
IndexError: bytearray index out of range

**GUI:**
Extracting FRF for unlock...

But if i try on another car with **SC8** / FL_8V0906259H__0001.frf
No problem at all:
Extracting FRF for unlock...
Starting to flash the following software components : 
FD_0 : 1 : CBOOT : SC8E0L20 : 
FD_1 : 2 : ASW1 : SC8E0O20 : 
FD_2 : 3 : ASW2 : SC8E0O20 : 
FD_3 : 4 : ASW3 : SC8E0O20 : 
UNLOCK_PATCH : 9 : PATCH_ASW3 :  : 
FD_4 : 5 : CAL : SC800O20 : 8V0906259H 
,....etc

Why i can't UNLOCK **SCG** on: FL_5G0906259Q__0005.frf

bri3d, 2023-12-21:
Did you select Simos 18.10 from the ECU Type dropdown?

Also the SHAsum should be:

```
shasum FL_5G0906259Q__0005.frf                                
12a36f1109b0bf6f5d9107761dd0fb47557fb675  FL_5G0906259Q__0005.frf
```

bri3d, 2024-09-25:
Closing this issue; if problems persist, attach full logs.

closed 2024-09-25 by bri3d