bri3d/VW_Flash #117: VW_Flash wont connect_Simos Tools doesnt work_MachinoA0 on 0.9

Mushlet, 2023-12-24:
Got a brand new Macchino A0 which has been flashed with Switch 0.9 firmware. That goes well.

SimosTools doesnt connect 
VW_Flash doesnt connect and has the following log...

Any ideas what is missing here? All help much appreciated

[image]

bri3d, 2023-12-24:
To use 0.9 with VW_Flash, you need to use J2534 rather than the direct connection functionality. The direct connection functionality is scheduled to be removed entirely in the future.

Mushlet, 2023-12-24:
So select the J2534 option instead?

That gives this...
[image]

Mushlet, 2023-12-24:
Doesnt really explain why simos tools also wont connect. 

Is there any data that i can collect that you may want to see to help? I appreciate your time

Mushlet, 2023-12-24:
Ive no idea how, other than reinstalling the full lot for about the 15th time.... but now it connects on both the simostools app and vw flash.

closed 2023-12-24 by Mushlet