bri3d/VW_Flash #120: Cannot unlock ECU - 8R2907115C 0002, box not found in .csv

kpematopa, 2024-01-24:
Hello,
I am trying to unlock ECU 8R2907115C 0002 but I got errors that the ECU is not known. Can we add it in the supported boxes?

kpematopa, 2024-01-24:
[image]

kpematopa, 2024-01-24:
[image]
[image]

kpematopa, 2024-01-24:
I unlocked the ECU, but my part number is not listed in [box_codes.csv](https://github.com/bri3d/VW_Flash/blob/master/data/box_codes.csv)
I guess I cannot use CAL file, but the full unlocked flash file .bin
Unlocking of ECU was 15 mins with FL_8V0906259H__0001.frf

bri3d, 2024-01-24:
Yes, the full file should work. I can add this file to box_codes.csv for next release; I was not aware of 8R2 (Q5 without FlexRay) box codes, only the Simos18.4 ones for which I never finished an unlocking patch.

kpematopa, 2024-01-24:
Mine is 18.1, model year 2015/2016, it is the first Q5 but facelift. Very rare engine with 8 injectors. When I flash full bin file do I need to calculate checksums?

bri3d, 2024-01-24:
Checksums should be calculated automatically by VW_Flash.

aarons3, 2024-01-24:
Very cool, I never knew that existed either.

Checksums will be corrected automatically (all blocks) by VW_Flash.

kpematopa, 2024-01-24:
Do you plan to add unlock patch of simos 18.41?

kpematopa, 2024-01-29:
Is it possible to avoid flash counter to increase on every flash or it is a function of the MCU?

bri3d, 2024-01-29:
re: 18.41, probably not, I have no car to test with and no flexray hardware to use for bench development. The port flashing patch should be quite simple (it's the same as the 18.10 one with some addresses changed) but testing it is very high risk. 

re: flash counter, it has nothing to do with the MCU, it's just a variable stored into a dflash channel. No personal plans to patch it though, warranty fraud isn't really my thing.

FirewinX, 2024-07-21:
8S0906259N is also not in the box_codes.csv is it possible to unlock with FL_5G0906259Q__0005.frf?

aarons3, 2024-07-21:
Yes, same exact process unlocking with the 5G090629Q_0005 frf