bri3d/VW_Flash #123: Request: Audi AL551 support

kippdipp, 2024-02-09:
New to GitHub so don’t know if there’s another place for comments / requests like this.. 
For “kicks and giggles” I connected my D4 Audi A8 4.2FSI 8HP55, Audi calls this transmission AL551, to VW_Flash since it uses same controller as DQ380/381 (Renesas SH72519/SH72549). Get info appears to be working and I have an A2L+Hex for this transmission I could share. What would it take to get VW_Flash working on the AL551?

Side note, CCP is enabled so I was able to use the upload command to dump 0x180000 -0x200000 and 0x600000-0x680000 without seed/key command. Couldn’t get any of the ASW addresses sadly.
[image]

bri3d, 2024-02-25:
The best way to figure out what's going on would be to look at an ODX. It will contain the SA2 Seed/Key and the update payload. It might be the same as DQ381 entirely (same seed/key and joke AES key/IV) or it could differ.

kippdipp, 2024-02-25:
Here’s the FRF for the 1006 software version on my AL551: https://race-tun.com/vag-sgo-frf/audi/fl4h1927158ad1006.frf/fl-4h1927158ad-1006.frfori/

aarons3, 2024-02-25:
I compared a common 0GC300012A_14XX file to what was uploaded...

SA2 bytefields are identical

DQ381 uses ENCRYPT-COMPRESS-METHOD "AA"

That AL551 file uses "22"

kippdipp, 2024-03-04:
> I compared a common 0GC300012A_14XX file to what was uploaded...
> 
> SA2 bytefields are identical
> 
> DQ381 uses ENCRYPT-COMPRESS-METHOD "AA"
> 
> That AL551 file uses "22"

How big of a change those this require? I'm not familiar with compress methods

ConnorHowell, 2024-03-05:
> > I compared a common 0GC300012A_14XX file to what was uploaded...
> > SA2 bytefields are identical
> > DQ381 uses ENCRYPT-COMPRESS-METHOD "AA"
> > That AL551 file uses "22"
> 
> How big of a change does this require? I'm not familiar with compress methods

A bench dump would be needed to reverse engineer whatever encryption/compression methods are used for the flash when flashing the TCU.

averagejoe8, 2024-04-04:
> New to GitHub so don’t know if there’s another place for comments / requests like this.. For “kicks and giggles” I connected my D4 Audi A8 4.2FSI 8HP55, Audi calls this transmission AL551, to VW_Flash since it uses same controller as DQ380/381 (Renesas SH72519/SH72549). Get info appears to be working and I have an A2L+Hex for this transmission I could share. What would it take to get VW_Flash working on the AL551?
> 
> Side note, CCP is enabled so I was able to use the upload command to dump 0x180000 -0x200000 and 0x600000-0x680000 without seed/key command. My main goal is to edit the ASW but I couldn’t dump any of the ASW addresses to sadly. ![1706935084953blob](https://private-user-images.githubusercontent.com/119031637/303544762-0e59997a-8caf-48b4-a0c3-560af129ba98.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIyMjM1MTYsIm5iZiI6MTcxMjIyMzIxNiwicGF0aCI6Ii8xMTkwMzE2MzcvMzAzNTQ0NzYyLTBlNTk5OTdhLThjYWYtNDhiNC1hMGMzLTU2MGFmMTI5YmE5OC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDRUMDkzMzM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDMwYTRhYTI4MmE4YzhmMDM2ODM4MmFkMGI3OWFjODE2MWZhMmYwMzI5ZDdmODMxYWFhMzUzOTNkNGJjZjRkMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.rLiLvjU5d-smgZhCtBedfqf0mTXEMQxHOo_9vbB75nA)

It will be nice if You can share hex+a2l for this TCU.

kippdipp, 2024-04-13:
> A bench dump would be needed to reverse engineer whatever encryption/compression methods are used for the flash when flashing the TCU.

Here's bench dump from my transmission.
[R5F72519R_bhid0301_1034420214_eeprom-20240413-150127.zip](https://github.com/bri3d/VW_Flash/files/14968464/R5F72519R_bhid0301_1034420214_eeprom-20240413-150127.zip)

kippdipp, 2024-12-06:
"AL551 decompression is 2047 sliding instead of the conventional 1023" if that helps...

kippdipp, 2024-12-09:
I was able to cross reference and figure out the memory segments for the AL551 with the SH72519, see below in A2L format:
      /begin MEMORY_SEGMENT
        Pst0 "" RESERVED FLASH INTERN 0x0 0x40000 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x0 0x0 0x40000
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x0 0x0 0x40000
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Pst50200 "" RESERVED FLASH INTERN_2 0x40000 0x80 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x40000 0x40000 0x80 
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x40000 0x40000 0x80 
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Pst502B0 "" CODE FLASH INTERN_1 0x40080 0x13FAF0 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x40080 0x13FAF0
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x40080 0x13FAF0
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Pst2DFD70 "" RESERVED FLASH INTERN_3 0x17FB70 0x490 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x17FB70 0x17FB70 0x490
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x17FB70 0x17FB70 0x490
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Dst2E0000 "" RESERVED FLASH INTERN_4 0x180000 0x280 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x180000 0x180000 0x280
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x180000 0x180000 0x280
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Dst2E02B0 "" CODE FLASH INTERN_2 0x180280 0xFD80 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x180280 0x180280 0xFD80
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x180280 0x180280 0xFD80
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Dst300000 "" DATA FLASH INTERN_1 0x190000 0x6FD60 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x190000 0x190000 0x6FD60
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x190000 0x190000 0x6FD60
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Dst3FFD60 "" RESERVED FLASH INTERN_5 0x1FFD60 0xA4 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x1FFD60 0x1FFD60 0xA4
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x1FFD60 0x1FFD60 0xA4
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Dst3FFE04 "" DATA FLASH INTERN_2 0x1FFE04 0x4 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x1FFE04 0x1FFE04 0x4 
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x1FFE04 0x1FFE04 0x4 
        /end IF_DATA
      /end MEMORY_SEGMENT
      /begin MEMORY_SEGMENT
        Dst3FFE08 "" RESERVED FLASH INTERN_6 0x1FFE08 0x1F8 -1 -1 -1 -1 -1 
        /begin IF_DATA ASAP1B_DIAGNOSTIC_SERVICES 
          ADDRESS_MAPPING 0x1FFE08 0x1FFE08 0x1F8
        /end IF_DATA
        /begin IF_DATA ETK 
          ADDRESS_MAPPING 0x1FFE08 0x1FFE08 0x1F8
        /end IF_DATA
      /end MEMORY_SEGMENT