bri3d/VW_Flash #124: Not finishing flashing HELP

Hanziie, 2024-03-27:
I'm getting this when trying to unclock ECU. result is the same with v0.6.0 and v0.5.4

Using tactrix (genuie). But i'm unsure if i need to install anything else from your windows downloads?

[image]
[flash.log](https://github.com/bri3d/VW_Flash/files/14780551/flash.log)
[flash_details.log](https://github.com/bri3d/VW_Flash/files/14780552/flash_details.log)
[udsoncan.log](https://github.com/bri3d/VW_Flash/files/14780553/udsoncan.log)

bri3d, 2024-03-28:
Your ECU is responding with an incorrect response to the request for a
Programming session. Is the immobilizer free and was the ECU already tuned
with another tool?

Hanziie, 2024-03-28:
Immo is not sync'ed  as this is a New(used) ecu with differens t power class

And I don't know if it's been tuned before

bri3d, 2024-03-28:
Without immo being unlocked, the ECU will never enter OBD programming. You
need to sync the immo online with ODIS or flash the ECU in boot mode.

Hanziie, 2024-03-28:
Can VW_FLASH Do this? I have a raspberry pi.

closed 2024-09-25 by bri3d