bri3d/VW_Flash #129: this bar and button are missing in windows11

Eddy0697, 2024-06-02:
this bar and button are missing in windows11

[image]

bri3d, 2024-06-02:
> this bar and button are missing in windows11
>
> Screenshot_20240603_011919_Gallery.jpg (view on web)

can you try manually resizing the window by dragging it? sometimes in the
past there have been wxpython bugs around the initial layout that are
solved by dragging the window to be larger

Eddy0697, 2024-06-02:
Yes, I've already pulled it bigger because I thought about it too, but nothing has changed.

aarons3, 2024-06-03:
Works fine on windows 11 here.  Do you have a screenshot of the issue?  Are you using the EXE version or native python?

Eddy0697, 2024-06-03:
Problem solved. I lowered the scaling for all apps.

bri3d, 2024-09-25:
Good solve. Closing out.

closed 2024-09-25 by bri3d