bri3d/VW_Flash #131: How to use VW Flash without a FRF

Golfather, 2024-06-07:
Hi chaps

For this sw version no FRF was ever released:

8S0906264
0002
2018 Audi TT CJSA 1.8T  Simos 18.1

I have found a .bin read by an unknown tool but not sure if its been encrypted and therefore no good to use with VW Flash.

Is there any way to check or parse the bin to ensure it is usable by VW Flash before i brick the ECU?!

https://mega.nz/file/5FJQmR5K#jlAfzkXN0epsmubinUsFfiYDbEWjxUn7j2IrNCco3bs

Thanks

aarons3, 2024-06-07:
I plopped the calibration block of this file into a known good SC8O40 bin
and they were identical.  ASW and bootloader blocks are definitely not
encrypted.  Obviously a little bit unknown original of this file, but it
looks good to go in terms of VW Flash compatibility.

Golfather, 2024-06-11:
Ok chaps, found an actual FRF for the same engine code CJSA except for its for an Audi A3
FL_8V0906264H_0005

All the CBOOT and ASW data is identical between the bin of 8V0906264H  and 8V0906264H
Only CAL areas are different, which is to be expected as different vehicles

Now more confident in unlocking this ECU as the bin for  8S0906264 seems legit
However the question now is if i need to relock the ECU, I will still need the original FRF

Is there any way we can recreate a FRF for 8S0906264_0002 using 8V0906264H_0005.frf with the CAL of 8S0906264 ?

aarons3, 2024-06-11:
There's no difference between using a bin or frf, be it to unlock the ECU or relock it / flash a totally stock file.  With that bin that you already have, you have all need to flash back to stock/relock if needed.

Golfather, 2024-06-11:
Ok thanks, thats useful to know, appreciate the help :)

Golfather, 2024-06-14:
Worked a treat. Thanks all !

closed 2024-06-14 by Golfather