bri3d/VW_Flash #134: Simos flash failed on "RequestSequenceError" - Engine won't start

nlis88, 2024-06-23:
I tried to unlock the ECU, but during the attempt to upload the unlocking file, the application freezes at a certain point. The logs show an error: RequestSequenceError. Repeated attempts to upload the file result in the same error. I used the Tactrix OpenPort interface for the upload. Please help as the car won't start.
[flash.log](https://github.com/user-attachments/files/15945419/flash.log)
[flash_details.log](https://github.com/user-attachments/files/15945420/flash_details.log)
[udsoncan.log](https://github.com/user-attachments/files/15945421/udsoncan.log)

bri3d, 2024-06-23:
Your lzss.exe isn’t in the right place or isn’t correct, or some weird
Windows file permissions thing is keeping it from compressing the block
content properly. Block compression is failing. Are you using a released
version of VW_Flash? If so which one?

nlis88, 2024-06-23:
It’s possible I messed up something with the versions. I downloaded version v0.7.0, but after extracting and launching the software, when I try to click "unlock" or "extract frf", errors about missing files pop up, e.g., FileNotFoundError for the file: frf.key. What I did was I copied only the VW_Flash_GUI.exe file into the cloned VW_Flash master repository and launched VW_Flash_GUI.exe from this location.

bri3d, 2024-06-23:
Please just download the latest stable release 0.6.0 for now and run it on
its own.

nlis88, 2024-06-23:
I repeated the entire process of unlocking and programming the ECU with version v0.6.0, and everything went smoothly. Thanks for the quick help.

nlis88, 2024-06-23:
Fixed using version v0.6.0

closed 2024-06-23 by nlis88