bri3d/VW_Flash #135: DQ381 FRF Full flash

Eddy0697, 2024-07-02:
Hello, flashing a calibration file for the dq381 works. but why doesn't flashing a full FRF work? stays at block 3, 100% and then nothing happens anymore. nothing was taken over

bri3d, 2024-07-02:
These control units have a bug in their reflash process where block erase
and validate requests fail and need to be retried indefinitely. I haven’t
worked around this bug yet. Best to full flash with ODIS-E or VCP if you
need to change versions.

johanbeerden, 2024-07-04:
> These control units have a bug in their reflash process where block erase and validate requests fail and need to be retried indefinitely. I haven’t worked around this bug yet. Best to full flash with ODIS-E or VCP if you need to change versions.
> […](#)

Does this apply to the newest (post 2022) DQ381's? Is there any way that clutch pressure and shifting points can be set without actually having to decrypt/crack the TCU?