bri3d/VW_Flash #136: DQ381 VW Golf R 7.5 stop flashing after flashing block: 2 from file FD_02DATA

IVANVXR, 2024-07-12:
Hello, I was try to back gearbox to stock via original .frf file and software stuck after Flashing block: 2 from file FD_02DATA.
After that I was use ODIS-E to flash same file and all is okay after.
Regards

[flash_details.log](https://github.com/user-attachments/files/16199411/flash_details.log)
[flash.log](https://github.com/user-attachments/files/16199412/flash.log)
[udsoncan.log](https://github.com/user-attachments/files/16199413/udsoncan.log)

IVANVXR, 2024-07-29:
Anyone?

bri3d, 2024-07-29:
This is expected at this time, the controller has a lot of bugs in the
flashing routine (ODIS-E just retries until it works) and I don’t have one
to test with. Only flash calibration on DQ381