bri3d/VW_Flash #139: Flashing stopped working after 3 or 4 successes, hangs/aborts at "Upgrading to programming session..."

jbhebert77, 2024-07-21:
Was able to unlock and successfully flash the ECU 3 or 4 times, but now when I try to flash it with either a tune or stock file, it gets to "Upgrading to programming session..." and aborts.  If I try to relock, I get the same error.  If I try to unlock again, I get an error unpacking the file.  "Get ECU Info" still works.  I suspect I did something I shouldn't have and bricked the ECU, but I want to know if there is any way to recover.

Logfiles too large to attach; archived here instead:

https://drive.google.com/drive/folders/1qJGLlSsSWACjGYETGZLhUOs9gCTjNi9B?usp=drive_link

bri3d, 2024-07-21:
Hi; I can’t see this Google Drive link. Usually this means you flashed a
file without a matching power class, but it could have a less fatal cause
too.

jbhebert77, 2024-07-24:
My apologies, permissions have been updated.

It was a Hail Mary pass at an attempt to get a CTUA block running with a CXCA ECM and sensors, so it wouldn't surprise me if that is the case. 

Thank you for your time.

arell50, 2024-09-05:
> Was able to unlock and successfully flash the ECU 3 or 4 times, but now when I try to flash it with either a tune or stock file, it gets to "Upgrading to programming session..." and aborts. If I try [...]

did you solve this issue?