bri3d/VW_Flash #144: Haldex Flashing struggles

KiethBurton69, 2024-09-25:
Hi,
 
I cant get the Haldex flasher to flash past block 1 using CLI in vs code python v3.11. 

I have a complete modified dump i need to flash but also cant get any FRF's to flash.  The driver block 0 will flash fine but it then gives up, have you got any ideas? I have a "dashing module not loaded" message at the top of the terminal however all other functions are working as intended i have comms with interface and controller. 

Log Files attached 
Thanks 
[flash.log](https://github.com/user-attachments/files/17129187/flash.log)
[flash_details.log](https://github.com/user-attachments/files/17129188/flash_details.log)

ConnorHowell, 2024-09-25:
Could you send over the files you're trying to flash? Preferably the first file you tried to flash? The log is showing as every single checksum being incorrect which doesn't sound right...

KiethBurton69, 2024-09-25:
Hi Connor, 

Here is the files, it gives the same checksum errors when flashing the mod. My plan was to flash the frf which is the same version on the controller currently to ensure a complete flash and then the mod. 

Thanks 

[FILES.zip](https://github.com/user-attachments/files/17129555/FILES.zip)

KiethBurton69, 2024-09-25:
I also tried removing the checksum checks as @threepotMR2 did in the bin.py file as both the odx and mod file dont need any checksum corrections i need it to flash untouched

KiethBurton69, 2024-09-25:
[udsoncan.log](https://github.com/user-attachments/files/17132434/udsoncan.log)
Looking at the uds log its returning a conditions not correct error before aborting, which is odd because the driver block meets conditions and flashes no problem. This is a 2019 S3 8V which requires the hood to be open while flashing. Have you had any issues like this on the later years?

threepotMR2, 2024-09-25:
I will add this, here is a dump of an actual real 0cq907554k sw7762 hw55 module.
Along with keiths ODX extracted with VCP into the seperate files.

It will not flash for me either, it fully completes the write, but then the module is bricked (party time!)
I've tried it twice.
I will dump a failed written until later tonight.
[0cq907554k sw7762 h55.zip](https://github.com/user-attachments/files/17133902/0cq907554k.sw7762.h55.zip)

KiethBurton69, 2024-09-25:
@threepotMR2  

Thanks for extracting it! Thats odd it actually fully flashed for you but bricked, did you leave checksum alone? Also ive just realised you will have the mod file please keep that to yourself as it was super expensive to obtain :)

threepotMR2, 2024-09-25:
Where is the mod from?

threepotMR2, 2024-09-25:
My two units are soft bricked because the last block of data has not been written, see image for comparison.

[image]

Here is the read back after brick file and the good file to compare in a hex editor if you wish.

[bricked data and good data.zip](https://github.com/user-attachments/files/17137857/bricked.data.and.good.data.zip)

I've messed with alot of these Gen 5 haldex files, and this 0cq907554k file is the only one I've seen so far with data after that 0x4DBF0 line with the checksum at the end.

Which makes me think there is something odd going on with the "FL_0CQ907554K__7762_.odx" which has been used as the source for the driver section etc. There is a line at 0xB400 which give the ID number of the source file, and that ODX you've been given states 0cq907554H_7083 which is a normal Golf or A3 or TT unit. But the 0cq907554K_7762 is the magic one from RS3 or TTRS with the harder tune and the advanced drive selection modes. I'm not sure the source of the file is being totally transparent.

[image]

bri3d, 2024-09-25:
I think that what's going on here is twofold: 

* This RS3 file has been tampered with in various ways to cross-flash back onto a Golf R successfully
* The VW_Flash Haldex support was written with only Golf R files as a base (you'll note this K version is not in the Flashdaten, so we never had it as a sample), so it does not understand this weird extra data after the "version" block. I think this is the issue with the Transporter files too.

Connor's going to look into fixing up the flasher to support variable-length blocks; no other control unit really does this thing where different versions are flashed as different sizes.

KiethBurton69, 2024-09-25:
Hi thanks for looking into the files just to let you know the odx supplied
and the mod file where not sold as the same version or anything. Simply the
car I have here was already on the 7762 version to begin with and I wanted
to flash a stock file to ensure the flash process worked before trying to
flash the modified software.

The modded file isn’t a 7083 the creator changed out all the version
identifiers to throw off the source of the real base file. I’m yet to find
the full matching version. However as you have noticed there is a block of
the 7762 file that matches the mod but the rest is different. I’m thinking
the base is a version close to 7762. I have a working controller with the
mod installed, I need to find a way to flash this file as is into other
controllers.

Thanks

threepotMR2, 2024-09-25:
There is only 1 table that has actually been modified in this file, and its a DTC/fault code table that's has some codes deleted out of it. 

[image]

So I'd say that is almost certainly correct that its the RS3 file fudged to run on the MQB's golf/a3/tt etc.
100% the source of this file is the 0CQ907554K__7762

ConnorHowell, 2024-09-25:
> There is only 1 table that has actually been modified in this file, and its a DTC/fault code table that's has some codes deleted out of it.
> 
> ![Screenshot at 2024-09-25 21-40-16](https://private-user-images.githubusercontent.com/171820636/370893486-5c7c3a0e-2052-4f92-af94-8dd54c205ab8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyOTgxNTQsIm5iZiI6MTcyNzI5Nzg1NCwicGF0aCI6Ii8xNzE4MjA2MzYvMzcwODkzNDg2LTVjN2MzYTBlLTIwNTItNGY5Mi1hZjk0LThkZDU0YzIwNWFiOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNVQyMDU3MzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYTliMTdhY2Q3MWJhNmZhNGIxZGEzNWE0YmIzOGY3YzQ4YzliOTZlNzg0OTVjZjBkYzhhMTBlYTMyZDAxMTllJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zrKrgpQdi7QhX_imoofhjE691XIBlc4zx1RZJww3058)
> 
> So I'd say that is almost certainly correct that its the RS3 file fudged to run on the MQB's golf/a3/tt etc. 100% the source of this file is the 0CQ907554K__7762

Yep this is definitely it, noticed that earlier! These are all U112200 codes so almost certainly just to facilitate the cross flash onto an RS3 file

KiethBurton69, 2024-09-25:
Oh wow that’s a very good point well spotted yeah I see what you mean. I
will get the car on the dyno tomorrow and do some testing between the two
controllers, I will be able to see if any more power is sent to the rear
which is the main wish from this software. Interestingly the 7762 version
is on this 2019 S3 I have here from factory, even though the module itself
is stamped 7084 on a J variant. So they must have started using the RS3
calibration on the last of the 8V models.

ConnorHowell, 2024-09-25:
@threepotMR2 @KiethBurton69 I've just done some really hacky changes to make it use variable lengths for the ASW & VERSION blocks on my fork (https://github.com/connorhowell/vw_flash)

Passing a bin file to either the CLI or GUI will now take the block lengths from the checksum header instead of assuming a fixed length.

KiethBurton69, 2024-09-25:
Well I’m glad I spent good money on it 😂 at least we know now!

threepotMR2, 2024-09-25:
@ConnorHowell I'll give it a blast now and see how many more I can brick today.

KiethBurton69, 2024-09-26:
@threepotMR2 how did you get on? I still cant get it to flash it does not even attempt the first block now just fails. Here are the logs.  Im using this prompt in the cli python VW_Flash.py --action flash_bin --input_bin "INSERT FILE" --interface J2534 --unsafe_haldex 

Tried using the gui also but it rejects the file as not being a valid bin or frf, does it make a difference that the mod file is saved in hex? I tried converting to bin also but thats the same outcome 

[flash.log](https://github.com/user-attachments/files/17145993/flash.log)
[flash_details.log](https://github.com/user-attachments/files/17145994/flash_details.log)
[udsoncan.log](https://github.com/user-attachments/files/17145995/udsoncan.log)

ConnorHowell, 2024-09-26:
> @threepotMR2 how did you get on? I still cant get it to flash it does not even attempt the first block now just fails. Here are the logs. Im using this prompt in the cli python VW_Flash.py --action [...]

Yep absolutely matters that it's a .hex, even converting that to a bin file won't put the data in the places that VW_Flash expects, I'll email you over your files in the correct format shortly.

threepotMR2, 2024-09-26:
exactly this ^^^^

"HALDEX MOD - Copy.bin" will never write with vw_flash because its file size and formatting isn't correct.

But worry not, this is 100% flash-able with a bit of playing.

I faied to write it last night as at the same time I went to my lab at 11pm last night Github had some problem and was down. But at some point later today I will have a play and get it working - but its two stages to get it working. Firstly we have to hope @ConnorHowell's testing purposes fixes to the flash tool which have been aptly called "[haldex hack]" actually work, and then that the file correctly checksums when it is written as your mod file has the checksums missing.

KiethBurton69, 2024-09-26:
@ConnorHowell sent me the files correctly formatted which did infact allow me to flash! The first flash i done was mod on to the already modded controller which worked perfect. I then flashed an frf to go to the controllers original version of 7083 to test if i could go from that back to the mod as if it was on a stock car, the frf flashed and worked fine but when flashing back to the mod it bricked. Wrote successfully though so the code is working!

threepotMR2, 2024-09-26:
Did you dump this hex file yourself from a control unit? Is it from a HPA one?

KiethBurton69, 2024-09-26:
I did yeah using a miniwriggler, it’s a unit that’s been flashed by an end
user tuning tool which we then dumped. It now makes sense that the company
who sold the software wasn’t keen to work with us because now we know their
secrets 😂

KiethBurton69, 2024-09-26:
@threepotMR2 how are you un bricking modules, using Odis or the miniwriggler?

ConnorHowell, 2024-09-26:
> @threepotMR2 how are you un bricking modules, using Odis or the miniwriggler?

Only the miniwiggler will allow you to unbrick (it's what you'd usually call a boot mode on commercial tools), you should just be able to flash the .hex file you have to get it back to a working state.

ConnorHowell, 2024-09-26:
@threepotMR2 I think you're going to need a bigger drawer soon lol

Interesting those aren't recoverable with a miniwiggler though...

threepotMR2, 2024-09-26:
If mess with some of the earlier memory blocks then the Jtag bootloader doesn't work. So be very very careful which blocks you choose to write with miniwiggler/memtool.

Not all those boards were bricked, some are just too badly corroded or burnt out.

If you are writing using vw_flash then you cannot hard brick a Haldex controller, it can only be soft bricked, because it seems like it stays away from erasing those blocks.

KiethBurton69, 2024-09-26:
Oh wow now that is impressive! Yeah luckily mine is soft bricked, could you please explain how to write with memtool so that i dont fully brick it lol, i just want to load the full hex dump back into it
[image]

threepotMR2, 2024-09-26:
You are better doing a full read, in box hex and bin format.... then post it here, then I'll direct you to only write those broken blocks

KiethBurton69, 2024-09-26:
Great thank you, here is a hex read however i cant find an option to read as bin 
[BRICKED AFTER FRF THEN MOD DUMP - Copy.zip](https://github.com/user-attachments/files/17152036/BRICKED.AFTER.FRF.THEN.MOD.DUMP.-.Copy.zip)

threepotMR2, 2024-09-26:
You need to write just the area C0B400 to C0EFFF

KiethBurton69, 2024-09-26:
> You need to write just the area C0B400 to C0EFFF
> 
> It still has the data between C0B410 to C0EFFF from the old "mod" flash.
> 
> e.g. FLASHDATA_Addr_0002.bin section never was written.

Thats odd because i wrote it will a full frf from version 7083 prior to flashing the mod which worked fine. Thanks ill get it going tomorrow :)

threepotMR2, 2024-09-26:
I've always wondered why if I try to erase/write the full chip with Memtool then it always stops connecting via JTAG.
Maybe this in the datasheet has something to do with it.... 

[image]

"DBGPRR value in Flash addr. C001F0 H"
"The last column of this table shows the value, which must be written into Flash location C001F0 H"

When I erase the block 0 being c00000 to c00FFF it fills that whole area with 00 bytes which means the DBGPRR value is 0000h being "Debug Interface = not available"

Block 0 upto end of to Block 8 being C08FFF contain the same data in every Gen 5 VAG haldex or VAQ ediff controller, it is the start of the internal program, its some kind of boot loader or something.

So I would guess **NEVER** **NEVER** **NEVER** try to erase Block 0 = C00000 to C00FFF
And Basically only flash block 9 onwards.

threepotMR2, 2024-09-27:
Also found this late last night in the document (xc27x4X_Errata_Sheet_v1.7.pdf) :-

[image]

KiethBurton69, 2024-09-27:
Well found! Okay this is a very noob question but i cant work out how to setup memtool to flash just that section could you explain please before i kill this controller lol

KiethBurton69, 2024-09-27:
> Here is a video showing how to program only a selection of blocks at a time.
> 
> https://www.youtube.com/watch?v=4pfTHcWgl7Y

Well thats made me feel stupid the second i saw you select the sections it clicked, thank you! Only issue im having now is that it wont begin erasing and fails. Reads out fine and the first 8 bytes are intact so should flash fine.

threepotMR2, 2024-09-27:
@gt-inno 

[image]

bri3d, 2024-09-27:
> "DBGPRR value in Flash addr. C001F0 H" "The last column of this table shows the value, which must be written into Flash location C001F0 H"
>
> When I erase the block 0 being c00000 to c00FFF it fills [...]

Have you tried using the HWCFG pins to pull the device into CAN or UART Bootstrap Loader mode rather than using debug (JTAG) to access Flash? Bootstrap Loader should _always_ work regardless of the state of Flash, and should allow these "hard bricked" units to be recovered.

threepotMR2, 2024-09-27:
I am confident the CAN bootstrap loader is still accessible, and will probably be a more straight forward method to manipulate the flash, as looking at the PCB config resistors its setup to be in CAN BSL from the factory. With a good tool we should be able to just plug in to the external haldex connector and do the CAN BSL.

If there is some easy to use software and some supported CAN interface you suggest I will happily buy it and play. I do not have any software tools or hardware to try the CAN bootstrap method.

bri3d, 2024-09-27:
> I am confident the CAN bootstrap loader is still accessible, and will probably be a more straight forward method to manipulate the flash, as looking at the PCB config resistors its setup to be in [...]

For CAN development I use a Raspberry Pi with a CAN hat and SocketCAN, programmed in Python.

I believe https://github.com/bri3d/TC1791_CAN_BSL/blob/main/bootloader.py#L303 is the same protocol for XC2000 as Tricore (XC2000 is basically our good enemy C167 with Tricore peripherals synthesized onto it).

But, of course, a C167-compatible BSL binary will need to be compiled. 

https://www.infineon.com/dgdl/ap1616410_XE16x_FlashViaCANBSL.pdf?fileId=db3a3043243b5f1701245769271d09cc covers this in depth.

It sounds like the Hitex debug tool HiTOP53-UConnect-XE164 can also do this using provided software, but it is probably more of an expensive pain in the ass to find all of this stuff than to write a little BSL binary like I did for Tricore (famous last words... C167 is so evil...).

threepotMR2, 2024-09-27:
I have been hunting for an Infineon Uconnect for about 2 years on and off. There is a proper infineon CAN flashing file driver for minimon that I found years ago, but the Uconnect usb dongle eludes me!

The dutch guy who is [vd Veer Engineering](https://www.vdveer-engineering.nl/) I think has managed to do CAN BSL stuff on these. But again he is good at programming like yourself - I am just a mere bodger.

threepotMR2, 2024-09-27:
The controller isn't perfect, it is a man in the middle attack. But he is a very clever guy to get as far as he has with that project! PCB design, embedded software, reverses engineering all the CAN analytics, and the phone app - that's pretty good going for one guy as a side project. The clutches will always get nailed if you use them outside of OEM application, Haldex is really designed to get a Yeti or a Tiguan out of the snow. Haldex is not a real 4wd system, if people want to play with high power 4wd they need a central differential not a clutched PTO.

threepotMR2, 2024-10-01:
Hello @ConnorHowell 

There is some other issue with VW_Flash that is a slight problem, so when you have an unsuccessful flash that correctly writes the C10000 area OK but there is an error in the C0B400 area, the haldex ECU will connect via UDS, it'll give you its part number, but it will not reply with fault codes or anything else.

When you try to do a flash, it always checks for fault codes, and when they are partially corrupted like this you cannot recover it with VW_Flash..... but I can write them with VCP instead luckily. I think we need VW_Flash fixing so it does not check the fault code status first.

@KiethBurton69 
Miniwiggler will not write those earlier blocks either for me on some units, but it will always write the C10000 and onwards. If that area is fixed then VCP will write a standard FRF file to the whole unit which get you the unit working again, and if Connor can can address that problem above then VW_Flash will be able to recover it too.

Also can you please email me the files formatted by Connor as I can't seem to get this file flashed with the files formatted by myself and its driving me nuts! I'm obviously doing something stupid.

Whoever make this file is a clever person, and has access to the SGO files that are not public! I'm very intrigued.

KiethBurton69, 2024-10-01:
Hi @threepotMR2 

I see! Recovery with VW Flash would be super handy as its a pain getting the module setup on the bench.
Send me your email and ill fire them over, i have access to early non public Sgo files if needed however they are not cheap.  I have been waiting on another test car to arrive to test Connors last tweak. On friday i managed to set fire to my miniwriggler by moving power pins and shorting it while live lol, new one is here now and i have another haldex module so can mess around getting the other going another day. Test car will be here Thursday but let me know if you have success flashing our modified file.

threepotMR2, 2024-10-01:
threepot@hotmail.co.uk

I am experimenting with xprog which is has support for this micro and I think it works over CAN.

ConnorHowell, 2024-10-01:
@threepotMR2 

Interesting, I believe it's only trying to erase codes but is probably causing the issue. Erasing codes is usually a prerequisite to flashing some ECUs which is why VW_Flash will try to do it. I'll push a commit to my branch shortly to skip that step for the Haldex; then once it's all confirmed working I just need to fix all my hacky code lol!

Regarding the RS3 file, an sgo file is what you'd normally have for pre-uds control modules. FRFs are what is used for anything modern, these are essentially encrypted zip files that contain an ODX file which contains the flash procedure and everything required to perform the flash (data, encryption/compression identifiers, checksums etc.)

To be honest it would be trivial to fix the checksums in a haldex file, update the flash checksums & data in the ODX and use that with VCP to flash a modded file.

threepotMR2, 2024-10-01:
This is killing me! It will not flash the modded.bin or stock.bin.

I always get the partial brick problem... comms but not properly....

[image]
[image]
[image]

here are my logs.... help!!

[flash.log](https://github.com/user-attachments/files/17215857/flash.log)
[flash_details.log](https://github.com/user-attachments/files/17215858/flash_details.log)
[udsoncan.log](https://github.com/user-attachments/files/17215859/udsoncan.log)

bri3d, 2024-10-01:
Try changing STMin_TX

KiethBurton69, 2024-10-01:
Hmm yeah that’s exactly what happened to me, are you trying Connor’s latest
revision?

threepotMR2, 2024-10-01:
@KiethBurton69  Yes, it never works for me...

@bri3d I've just changed STMIN_TX to 200 and its done the same. I'll try it at 400 instead.

But I don't think its a hardware issue, I think its something screwed up with the variable size block 4 data.

ConnorHowell, 2024-10-02:
Wasn't an stmin issue, I was only half overriding the block lengths. @threepotMR2 try the latest change on my fork.

threepotMR2, 2024-10-02:
@ConnorHowell It works!

jakka351, 2025-10-28:
Info on XC2000 series CAN BSL here https://www.infineon.com/row/public/documents/10/42/ap1616410-xe16x-flashviacanbsl.pdf