bri3d/VW_Flash #145: Will VW FLASH work to recover immo locked Simos18.1?

em1ter, 2024-09-30:
Hello everyone,

First of all, thank you for this tool and project!

Sorry if my question annoys anyone but I'm desperately chasing some help in recovering of my ECM. Due to my own reckless actions I locked ECU in my car with flashing a wrong firmware with ODIS E.
ECU stock firmware: 3G0906259B 0002 (06K907425B H13)
I started with flashing with 8S0906259B 0005 with ODIS. All went well but after I found that ACC stopped working I decided to try another firmware.
So I flashed 8V0906259J 0003. And that went wrong. After bouncing the ignition and clearing all DTC I found P157000 "ECM locked/ECM deactivated" code in ECM. And now I'm unable to flash any other firmwares as ODIS simply cancels the flashing process.

My suspicion is that I have flashed a firmware that is not compatible with immo data stored in EEPROM. My hope is that VW FLASH will allow me to flash a compatible firmware regardless to ECM. The ECM was stock before my actions - no tuned and/or locked software, a complete factory stock that was working well in my car.

So here are my questions:
1) Will VW FLASH allow me to flash 3G0906259 firmware to ECM that is deactivated by immo data mismatch?
2) Does anyone know will this approach with flashing a stock firmware help to recover the ECU?
3) Are the actions described above harmful for existing immo data?

I have a "good" 5054 clone and I have a genuine Tactrix Openport 2.0. Will this set be sufficient enough to perform required actions?

Any help or advice is highly appreciated!

Thanks.

em1ter, 2024-09-30:
[bri3d](https://github.com/bri3d),
Thank you for such rapid and detailed response!

Just a notice that confirms your words: Initially the engine code was CJXA (206kW). After I flashed 8S0...B firmware it changed to CJXB (the same 206kW). But after I flashed 8V0...J it changed to CJXE (198kW).

The project you have provided is a priceless treasure! Correct me if I'm wrong, but as far as I understand with the guide provided and tools and docs in Simos18_SBOOT project it is possible to read and write with bench connection even immo locked ecu?

bri3d, 2024-10-01:
> But in my case, as far as I understand, it will not be enough to just flash the 3G0 firmware but will require DFLASH to be altered as it somehow got modified by 8V0...J firmware, is it correct?

No, the IMMO section of DFLASH should not have been modified. Flashing a calibration which has a match for the PClass byte in the IMMO data should unbrick the ECU.

> with the guide provided and tools and docs in Simos18_SBOOT project it is possible to read and write with bench connection even immo locked ecu?

Yes. Probably 8-10 people (that I know of) have been successful in using those instructions, but it's pretty involved and complicated - it was really intended as more of a research tool than an end-user thing.

em1ter, 2024-10-02:
[bri3d](https://github.com/bri3d),
Thank you for describing the possible path and solution.

I'm really keen to try SBOOT workaround.
However for this particular case I lack time and since not having this done previously I consider the risk to brick the ECM completely by doing something wrong with DFlash and PFlash to be extremely high. And if something goes wrong I will no longer have original immo data from my ECM so will have to buy a new ECM and have it coded because my local dealer does not perform replacements with used parts and refuses to perform their immo adaptation online. And no local independent workshops to carry out such works. Hence the decision was made to send my ECM overseas to a professional to perform a bench recovery.
And once it is done it is agreed that I will be provided with a full read from my ECM before and after the recovery works. As I'm eager to work out this case by myself I will get a donor ECM for experiments and make a clone with the full read from my bricked ECM - just to work out the issue in the manner you suggested.

Once again, thanks for your help!

em1ter, 2024-11-08:
@bri3d ,
Sorry to bump this thread but I'm not sure if there is any other way to communicate with you and receive your response.

After a plenty of time and a plenty of money spent on 'professional' work to recover my ECM by some other guy I've got it back in the same state I sent to him. Hence I decided to try to restore it by myself and have gone with your TC1791_CAN_BSL project. I have assembled the same tool you described (RPi 3B+ with Seeed CAN FD hat and level shifters), installed the project and all prerequisites, spent some time on resolving hidden dependencies that were causing bootloader.py to terminate with errors at different steps and it seems it finally has started working. Well, until I try to extract boot passwords. At the stage "Calculating key for seed:" script terminates with error: 
  File "/tools/simos18_tools/TC1791_CAN_BSL/./bootloader.py", line 271, in sboot_login
    print(sboot_seed.hex())
AttributeError: 'bool' object has no attribute 'hex'

I have searched in issues of TC1791_CAN_BSL projects and found a matching issue with the same exact problem, which was not responded. I have left my comment there as well.

Really hope you could find some time to work this out. This project is my only chance to get my car running again.
Thanks.

em1ter, 2025-01-11:
@arell50 ,
Yes, after a plenty of tweaking around with scripts and Brian's and others help I was able to make my RPi to work as BSL flasher and recovered my ECU with Brian's step by step guide described in TC1791_CAN_BSL project as well as used VW Flash to flash the ECU with factory firmware on the final step of the recovery process.

miliskot, 2025-04-04:
> Hello everyone,
> 
> First of all, thank you for this tool and project!
> 
> Sorry if my question annoys anyone but I'm desperately chasing some help in recovering of my ECM. Due to my own reckless actions I locked ECU in my car with flashing a wrong firmware with ODIS E. ECU stock firmware: 3G0906259B 0002 (06K907425B H13) I started with flashing 8S0906259B 0005 with ODIS. All went well but after I found that ACC stopped working I decided to try another firmware. So I flashed 8V0906259J 0003. And that went wrong. After bouncing the ignition and clearing all DTC I found P157000 "ECM locked/ECM deactivated" code in ECM. And now I'm unable to flash any other firmwares as ODIS simply cancels the flashing process.
> 
> My suspicion is that I have flashed a firmware that is not compatible with immo data stored in EEPROM. My hope is that VW FLASH will allow me to flash a compatible firmware regardless to ECM. The ECM was stock before my actions - no tuned and/or locked software, a complete factory stock that was working well in my car.
> 
> So here are my questions:
> 
> 1. Will VW FLASH allow me to flash 3G0906259B firmware to ECM that is deactivated by immo data mismatch?
> 2. Does anyone know will this approach with flashing a stock firmware help to recover the ECM?
> 3. Are the actions described above harmful for existing immo data?
> 
> I have a "good" 5054 clone and I have a genuine Tactrix Openport 2.0. Will this set be sufficient enough to perform required actions?
> 
> Any help or advice is highly appreciated!
> 
> Thanks.

Hi, can I ask if you were able to start it after flashing the ECU from 3G0906259B to 8S0906259B. It shouldn't work because there is a different power class... The stock one has 206 kW and the other one has 228 kW. Which shouldn't work... correct me if I'm wrong... the non-functioning ACC can be solved by uploading newer software via ODIS E FRF.

arell50, 2025-04-05:
> Hi, can I ask if you were able to start it after flashing the ECU from 3G0906259B to 8S0906259B. It shouldn't work because there is a different power class... The stock one has 206 kW and the other [...]

> Hi, can I ask if you were able to start it after flashing the ECU from 3G0906259B to 8S0906259B. It shouldn't work because there is a different power class... The stock one has 206 kW and the other [...]

The only way to recover this is to write fullflash by boot. You will need Magicmotorsports Flex or bitbox SIMOS18.x boot module to write boot file.