bri3d/VW_Flash #150: Simos 18.1 Unlock Fail "Attribute error ´FlashInfo´ object has no attribute ´patch_box_code´

LucaIP, 2024-12-14:
Hi
I have flashed a simos 18.1 with an FRF file from **FL_5G0906259E_005** to **FL_5G0906259F_002** , that worked so far. But when I try to unlock the ECU (using the .FRF file) an error always occurs. Can someone help me?
here the logs. the Programm always says the error: "extract the frf....." but i do. And in C the Error: "Attribute error ´FlashInfo´ object has no attribute ´patch_box_code´
Thx! 

[flash.log](https://github.com/user-attachments/files/18135065/flash.log)
[flash_details.log](https://github.com/user-attachments/files/18135066/flash_details.log)
[udsoncan.zip](https://github.com/user-attachments/files/18135068/udsoncan.zip)

bri3d, 2024-12-15:
You're using the wrong FRF file for Unlocking. Read the documentation again.

closed 2024-12-15 by bri3d

LucaIP, 2024-12-15:
Ich habe die Anleitung mehrfach durchgelesen. Wenn ich die falsche FRF benutze, welche sollte ich dann benutzen ?